arduino/Arduino #5098: IDE lacks "--help", "--version", man page

archenemies, 2016-07-04:
I don't know how recently the ability to use the Arduino IDE from the command line with arguments like `--upload` and `--port` was introduced, but this is very helpful. I think the next step is to document these arguments in a `--help` message and/or in a manual page. The ability to get the IDE version with a `--version` command would also be useful.

P.S. I can find no documentation which tells me how to specify the board on the command line. It would be nice to have such documentation. Somewhere it says I should specify "--board package:arch:board" but I don't know where to get "package", "arch", and "board" for a given selection in the "Boards Manager". At least an example would help.

P.P.S. See also http://arduino.stackexchange.com/questions/15893/how-to-compile-upload-and-monitor-via-the-linux-command-line/24802#24802

q2dg, 2016-07-04:
Well, there is https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc

archenemies, 2016-07-04:
Thank you, that's interesting. I remember seeing that page before and then I forgot about it.

So... I don't see this file in my `arduino-nightly` tarball. any idea why?

I guess I need to tell the folks who made my [arduino AUR package](https://aur.archlinux.org/packages/arduino/) that the manpage should be installed in a standard location; but first I should tell them where to find it.

(Also it looks like the AUR package is maintainerless, as of today...!)

nicrip, 2016-07-18:
Is there a list of supported boards for the --board option? I am using a Pro 3.3V 8MHz Atmega328.

VictorLamoine, 2017-03-16:
The man page is outdated:

```
ARDUINO(1)                                                                              General Commands Manual                                                                              ARDUINO(1)

NAME
       arduino - an IDE and uploader for Arduino development boards.

SYNOPSIS
       arduino

DESCRIPTION
       This program takes no options and will load the IDE in an X11 window.

SEE ALSO
       http://www.arduino.cc/

AUTHOR
       arduino was written by <developers@arduino.cc>.

       This manual page was written by Scott Howard <showard314@gmail.com>, for the Debian project (and may be used by others).

                                                                                             April 10, 2010                                                                                  ARDUINO(1)
 Manual page arduino(1) line 1/20 (END) (press h for help or q to quit)

```

VictorLamoine, 2017-03-17:
Ah right, I installed the latest Arduino release but I didn't realize the man page I was reading was from the Ubuntu repositories (= old version).

Sorry for the useless comment!

per1234, 2020-09-25:
`--version` option was added by https://github.com/arduino/Arduino/pull/7549.

The requested `--help` option is still missing.