arduino/Arduino #5099: osx IDE trackpad scroll inertia bug

tablatronix, 2016-07-04:
When using apple trackpad and scroll inertia is on, when you scroll to last line in editor, inertia scrolling causes the entire edit scrollspace area to bounce up and down until it settles.

oqibidipo, 2016-07-09:
Which IDE & OS versions? I don't see anything like that with IDE 1.0.6, 1.6.5 or 1.6.9 on OS X 10.11.5.

tablatronix, 2016-07-09:
latest version of both

tablatronix, 2016-07-09:
Looks like it only does it when there is no horizontal scroll bar also.

tablatronix, 2016-07-09:
I am using external touchpad, let me try the builtin one also/
I have a video gotta throw it somewhere

agdl, 2016-08-01:
I'm closing this for lack of feedback. Reopen it if needed

tablatronix, 2016-08-01:
I stopped using the ide, forgot all about this. Using another ide instead. Ill do some more testing , must be peculiar to my resolution or something if noone can reproduce.

tablatronix, 2016-08-01:
https://youtu.be/l3JUVOGfZI8

tablatronix, 2016-08-01:
re-tested on 1.6.10 and 10.11.6 again, still the same

oqibidipo, 2016-08-02:
Now I can reproduce it. Weird. 

> Looks like it only does it when there is no horizontal scroll bar also.

Same here.

tablatronix, 2016-08-02:
So can this be reopened or do it need to create a new issue?

hashier, 2016-09-11:
Have the same problem

z3t0, 2016-09-25:
Yes this seems to be a very annoying bug.