arduino/Arduino #5101: TransistorMotorControl example code issues

edwardmeshuris, 2016-07-05:
Hi,
I'm not sure if this is the right place to post this, but there are a few errors on the following page: [https://www.arduino.cc/en/Tutorial/TransistorMotorControl](https://www.arduino.cc/en/Tutorial/TransistorMotorControl)

There are two types of loops that are introduced on the page:

> ```
for(int x = 0; x < 255; x+5){
    analogWrite(motorControl, x);
    delay(50);
}
```

> ```
for(int x = 255; x > 0; x-5){
    analogWrite(motorControl, x);
    delay(50);
}
```

Here x is 'supposed' to be stepped by 5, until it hits the ceiling of 255 or the floor of 0. 

Issues:

1. The ceiling and floor (255 or 0 respectively) are never hit and the syntax for stepping is incorrect, should be re-written as:

  - `for(int x = 0; x <= 255; x+=5){`

 - `for(int x = 255; x >= 0; x-=5){`

The other loops are about the same, but are stepped by 1:

>```
// ramp up the motor speed
for(int x = 0; x < 255; x++){
    analogWrite(motorControl, x);
    delay(50);
}
```

>```
// ramp down the motor speed
for(int x = 255; x > 0; x--){
    analogWrite(motorControl, x);
    delay(50);
}
```
Issue:

1. The ceiling and floor (255 or 0 respectively) are never hit, should be re-written as:

  - `for(int x = 0; x <= 255; x++){`

 - `for(int x = 255; x >= 0; x--){`

I hope this is forwarded somewhere, the sketch is actually pretty neat, but a person new to this will be confused as to why the motor does not shut off at the end of the spin down.

agdl, 2016-07-07:
Done thank you!

closed 2016-07-07 by agdl

emeshuris, 2016-07-07:
Almost 🙃
`for(int x = 0; x <= 255; x+5){`
Should be:
`for(int x = 0; x <= 255; x+=5){`

Same for the decrement line.

agdl, 2016-07-07:
Sorry you are right. Done