arduino/Arduino #5103: Possible bug in Function declaration

Gorkde, 2016-07-06:
Please see my last post here:

http://forum.arduino.cc/index.php?topic=411069.0

When declaring a function after loop() it works but in this case (providing an init value in my script) it doesn't (not declared in this scope).

facchinm, 2016-07-06:
The issue you are encountering should have been mitigated by https://github.com/arduino/arduino-builder/commit/81eadf324ef2f3d14034452bd40bd2f2eb75ff98 , which was merged in the nightly some days ago. Would you mind giving it a try? https://www.arduino.cc/en/Main/Software#hourly

Gorkde, 2016-07-06:
Thanks for your quick reply.
I'm far behind with my project and can't take the time for installing and reinstalling. Can this be installed additional to the normal IDE or do I need to replace it?

facchinm, 2016-07-06:
You don't need to replace anything, simply unzip the package in a different location and launch the IDE from there :smile:

Gorkde, 2016-07-08:
Tried with Hourly 2016/07/05 9:33

Same result. Not declared....

matthijskooijman, 2016-07-08:
@facchinm, I don't think that commit will solve this issue. The problem is that the prototype generator does not understand default arguments, so it refrains from generating a prototype for such a function. This is probably a wontfix, since properly parsing default arguments is quite tricky to do right. However, it does result in unexpected behaviour for users, where adding a default argument results in a un-understandable error for a function that worked fine before.

I've previously wondered if we shouldn't stop generating prototypes on the longer term, but for a novice user with a simple program, the gains are significant.

I wonder if it would be good to:
 - Emit a warning when no prototype is generated for a function because of default arguments or other complexities (no warning is needed when an explicit prototype is already present)
 - Have some documentation on arduino.cc that explains what function prototypes are, how C/C++ normally need them and how the Arduino IDE autogenerates them, that can be linked to from the warning.

facchinm, 2016-07-08:
Hi @matthijskooijman ,
I read the issue without much attention and thought it was related with the <vector> like bug, my fault.
I agree so much with your proposal to emit a Warning when a function prototype is not generated, maybe adding a human readable reason (I believe the code is already in place and appears with debug-level = 10)

@Gorkde , sorry for the useless download, as Matthijs said you need to declare your function before its usage, such as

```
ShiftRegister74HC595 ShiftRegOut (1, 2, 4, 5); // Schieberegister Object erstellen (Anz. Register, Datenpin, Schiebetaktpin, Speichertaktpin (latch-pin)
void FadeOutput(int Pin, int StartWert, int EndWert, int Zeit = 1);

void setup()
{
....
}

void FadeOutput(int Pin, int StartWert, int EndWert, int Zeit)
{
...
}
```

Being a known limitation I'm closing the issue and I'm opening an issue on arduino-builder repo to talk about  the warning message

closed 2016-07-08 by facchinm

matthijskooijman, 2016-07-08:
@facchinm, will you create an issue at arduino-builder to track the need for the warning and extra documentation?

facchinm, 2016-07-08:
Yep, here https://github.com/arduino/arduino-builder/issues/167 :wink: