arduino/Arduino #5105: Freezing and Rebooting of Arduino

euvenilist, 2016-07-06:
Please help!
I am using Arduino GSM sheild with arduino Nano FTDI, also displaying a current change in 16x2 LCD, by making use of CT. 
my code:
void loop()
{
  float Itot =0,current=0.0;
    DisplayCurrent(double(calcIrms()),0,1); //Displaying current in a LCD
    if(sms.available())
    {
      sms.remoteNumber(senderNumber, 20); // get the sender number
      sms.beginSMS(senderNumber);
      sms.print(double(calcIrms())); // send the current
      sms.endSMS();
      delay(131);
      DisplayCurrent(double(Irms),0,1);
      sms.flush();  // Delete message from modem memor
     }
    delay(900);
  
}

1. Freezing problem: Sometimes when it reach the   if(sms.available())  it freez, it seems it keeps on running on the function and never come back to the loop().   
2. Rebooting problem: evertime I send an sms to GSM shield via my mobile, it reboot.

Please help.
Thanks in advance

Mpilo

Chris--A, 2016-07-06:
This is an issue for the Arduino forum: https://forum.arduino.cc/ not the IDE bug tracker.

Restart issues are caused by logic errors. 
You may be using too much RAM (Stack-Heap clash) or are accessing a dangling pointer, using arrays outside of their range, using undefined behavior (`i++ = array[i++]`) , and many other possibilities... 

But you can guarantee there is a logic error in there.

matthijskooijman, 2016-07-07:
Agreed with @Chris--A here, this is not something for this bug tracker. Most likely a logic error in your sketch or the library you are using, but it might also be caused by insufficient power perhaps.

I'm closing this issue. If it turns out this is an Arduino bug after all, feel free to reopen and add additional details about what happens exactly.

closed 2016-07-07 by matthijskooijman