arduino/Arduino #5106: Invalid UTF-32 Charakter

xpix, 2016-07-07:
I try to start the new IDE 1.6.9 and get this error message at the console:

`Loading configuration...
Initializing packages...
java.io.CharConversionException: Invalid UTF-32 character 0x7099e31a(above 10ffff)  at char #1552, byte #6211)
        at com.fasterxml.jackson.core.io.UTF32Reader.reportInvalid(UTF32Reader.java:189)
        at com.fasterxml.jackson.core.io.UTF32Reader.read(UTF32Reader.java:150)
        at com.fasterxml.jackson.core.json.ReaderBasedJsonParser.loadMore(ReaderBasedJsonParser.java:153)
        at com.fasterxml.jackson.core.json.ReaderBasedJsonParser._skipWSOrEnd(ReaderBasedJsonParser.java:2017)
        at com.fasterxml.jackson.core.json.ReaderBasedJsonParser.nextToken(ReaderBasedJsonParser.java:577)
        at com.fasterxml.jackson.databind.ObjectMapper._initForReading(ObjectMapper.java:3776)
        at com.fasterxml.jackson.databind.ObjectMapper._readMapAndClose(ObjectMapper.java:3721)
        at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:2796)
        at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:180)
        at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:93)
        at processing.app.BaseNoGui.initPackages(BaseNoGui.java:599)
        at processing.app.Base.<init>(Base.java:295)
        at processing.app.Base.guardedMain(Base.java:222)
        at processing.app.Base.main(Base.java:135)
`

Please can you kindly check this?
Thanks
Frank

facchinm, 2016-07-07:
Hi Frank,
do you have any additional json specified in "Additional Board manager URL"? If so, try removing it from your `preferences.txt` under Arduino15 folder

matthijskooijman, 2016-07-07:
Regardless of the cause, this parsing error should be caught and result in a proper error message, including the name of the invalid file.

xpix, 2016-07-07:
Thanks for catch this ... i removed complete the Arduino15 dir and it works.

agdl, 2016-07-08:
@xpix I'm closing this since it seems fixed. Reopen if needed

closed 2016-07-08 by agdl

dikenz, 2016-11-14:
Hey,

I am having the same in MacOS Sierra 10.12.1 with ide versions 1.6.9 and 1.6.12
after i added http://arduino.esp8266.com/stable/package_esp8266com_index.json  as suggested in 
https://learn.adafruit.com/adafruit-feather-huzzah-esp8266/using-arduino-ide

I am not able to start the IDE anymore using the same OS user. Deleting the folder does not help even deleting ~/Documents/Arduino doesn't.

Here is the exception:
java.io.CharConversionException: Invalid UTF-32 character 0x723a0200(above 10ffff)  at char #1552, byte #6211)
	at com.fasterxml.jackson.core.io.UTF32Reader.reportInvalid(UTF32Reader.java:189)
	at com.fasterxml.jackson.core.io.UTF32Reader.read(UTF32Reader.java:150)
	at com.fasterxml.jackson.core.json.ReaderBasedJsonParser.loadMore(ReaderBasedJsonParser.java:153)
	at com.fasterxml.jackson.core.json.ReaderBasedJsonParser._skipWSOrEnd(ReaderBasedJsonParser.java:2017)
	at com.fasterxml.jackson.core.json.ReaderBasedJsonParser.nextToken(ReaderBasedJsonParser.java:577)
	at com.fasterxml.jackson.databind.ObjectMapper._initForReading(ObjectMapper.java:3776)
	at com.fasterxml.jackson.databind.ObjectMapper._readMapAndClose(ObjectMapper.java:3721)
	at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:2796)
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:180)
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:93)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:599)
	at processing.app.Base.<init>(Base.java:295)
	at processing.app.Base.guardedMain(Base.java:222)
	at processing.app.Base.main(Base.java:135)