arduino/Arduino #5107: build folder only based on sketch name + path

tbowmo, 2016-07-07:
the build path is getting created in

https://github.com/arduino/Arduino/blob/master/arduino-core/src/processing/app/BaseNoGui.java#L132

And is based solely on the path and name of the sketch to compile. If we use arduino in a CI system, and want to do things in parallel for multiple archs, then they will collide with each other.

Can we have a bit more unique build path created? By adding timestamp for example in the string that get's md5ifized?

facchinm, 2016-07-13:
Hi @tbowmo,
the fixed path (obtained in the way you described) is useful to deterministically cache the build's intermediate results, speeding up the compile phase on the subsequent runs. 

Adding an arch:board field to the md5ified string would be a good solution, while timestamping would make the folder non deterministic.
Would you mind opening a PR with the first solution? would it be ok for your usage?

tbowmo, 2016-07-13:
Sounds reasonable.. was just easier for me to get timestamp, when doing initial experimenting. submitted PR #5130

matthijskooijman, 2016-07-13:
IIRC I have some pending patches that remove the determinism again, since it's not strictly needed. I'll see if I can find it again and what the status is.

tbowmo, 2016-07-13:
As long as we can have multiple builds on the same sketch running in parallel for multiple architectures, then it would be great..

matthijskooijman, 2016-07-14:
Ah, found it. It's the "Randomize the the build path name" commit in #4263. That PR is scheduled to be merged soon after 1.6.10 is released, last I heard.

matthijskooijman, 2016-11-23:
W00ps, linked the wrong PR, should have been #4363. Since that PR (in particular commit 59b70c83730126ea13b663c1114a58d30bcfe6b9) is merged, build paths are now randomized. @tbowmo, could you verify that your usecase is indeed covered now, and if so close this issue? The latest release version (1.6.13) should include this change.

tbowmo, 2016-11-25:
Verified to be ok..

closed 2016-11-25 by tbowmo