arduino/Arduino #5108: Boards.txt - select correct bootloader based on two menu selection

MCUdude, 2016-07-08:
Hi!
I'm creating an Arduino core for making it simple to use all the "non p" versions of the common ATmegas (ATmega328, ATmega168, ATmega88). My thought was that the user selects the microcontroller in the boards menu, and then close what clock frequency and variant (328p or 328). when the user hit "Burn bootloader", the correct bootloader should be chosen based on the options that's done. Note that the ATmega328 and 328p are using different bootloader versions to deal with the signature issue.

This is what it looks like:
[image]

How can I make the IDE choose the correct bootloader depending on not only one, but two options?
This is what the boards.txt file looks like right now. The bootloader `optiboot_flash/atmega328/optiboot_flash_atmega328_115200_16MHz.hex` will be chosen for both the ATmega328 and the 328p. I want the IDE to choose `optiboot_flash/atmega328p/optiboot_flash_atmega328p_115200_16MHz.hex` when the ATmega328p is selected. How can I do this? Is it even possible?

```
menu.clock=Clock
menu.BOD=B.O.D
menu.variant=Variant
############################################################

328.name=ATmega328
328.upload.tool=arduino:avrdude
328.upload.protocol=arduino
328.upload.maximum_size=32256
328.upload.maximum_data_size=2048

328.bootloader.tool=arduino:avrdude
328.bootloader.unlock_bits=0x3f
328.bootloader.lock_bits=0x0f

328.build.core=arduino:arduino
328.build.board=AVR_ATmega328
328.build.variant=standard

328.menu.variant.modelP=328P / 328PA
328.menu.variant.modelP.build.mcu=atmega328p
328.menu.variant.modelNone=328 / 328A
328.menu.variant.modelNone.build.mcu=atmega328

328.menu.BOD.2v7=2.7v
328.menu.BOD.2v7.bootloader.extended_fuses=0xfd
328.menu.BOD.4v3=4.3v
328.menu.BOD.4v3.bootloader.extended_fuses=0xfc
328.menu.BOD.1v8=1.8v
328.menu.BOD.1v8.bootloader.extended_fuses=0xfe
328.menu.BOD.disabled=Disabled
328.menu.BOD.disabled.bootloader.extended_fuses=0xff

#make atmega328 AVR_FREQ=16000000L BAUD_RATE=115200 UART=0 LED=B5 LED_START_FLASHES=2
328.menu.clock.16MHz_external=16MHz external
328.menu.clock.16MHz_external.upload.speed=115200
328.menu.clock.16MHz_external.bootloader.low_fuses=0xf7
328.menu.clock.16MHz_external.bootloader.high_fuses=0xd6
328.menu.clock.16MHz_external.bootloader.file=optiboot_flash/atmega328/optiboot_flash_atmega328_115200_16MHz.hex
328.menu.clock.16MHz_external.build.f_cpu=16000000L

#make atmega328 AVR_FREQ=20000000L BAUD_RATE=115200 UART=0 LED=B5 LED_START_FLASHES=2
328.menu.clock.20MHz_external=20MHz external
328.menu.clock.20MHz_external.upload.speed=115200
328.menu.clock.20MHz_external.bootloader.low_fuses=0xf7
328.menu.clock.20MHz_external.bootloader.high_fuses=0xd6
328.menu.clock.20MHz_external.bootloader.file=optiboot_flash/atmega328/optiboot_flash_atmega328_115200_20MHz.hex
328.menu.clock.20MHz_external.build.f_cpu=20000000L

#make atmega328 AVR_FREQ=12000000L BAUD_RATE=57600 UART=0 LED=B5 LED_START_FLASHES=2 
328.menu.clock.12MHz_external=12MHz external
328.menu.clock.12MHz_external.upload.speed=57600
328.menu.clock.12MHz_external.bootloader.low_fuses=0xf7
328.menu.clock.12MHz_external.bootloader.high_fuses=0xd6
328.menu.clock.12MHz_external.bootloader.file=optiboot_flash/atmega328/optiboot_flash_atmega328_57600_12MHz.hex
328.menu.clock.12MHz_external.build.f_cpu=12000000L

#make atmega328 AVR_FREQ=8000000L BAUD_RATE=38400 UART=0 LED=B5 LED_START_FLASHES=2 
328.menu.clock.8MHz_external=8MHz external
328.menu.clock.8MHz_external.upload.speed=38400
328.menu.clock.8MHz_external.bootloader.low_fuses=0xf7
328.menu.clock.8MHz_external.bootloader.high_fuses=0xd6
328.menu.clock.8MHz_external.bootloader.file=optiboot_flash/atmega328/optiboot_flash_atmega328_38400_8MHz.hex
328.menu.clock.8MHz_external.build.f_cpu=8000000L

#make atmega328 AVR_FREQ=8000000L BAUD_RATE=38400 UART=0 LED=B5 LED_START_FLASHES=2 
328.menu.clock.8MHz_internal=8MHz internal
328.menu.clock.8MHz_internal.upload.speed=38400
328.menu.clock.8MHz_internal.bootloader.low_fuses=0xe2
328.menu.clock.8MHz_internal.bootloader.high_fuses=0xd6
328.menu.clock.8MHz_internal.bootloader.file=optiboot_flash/atmega328/optiboot_flash_atmega328_38400_8MHz.hex
328.menu.clock.8MHz_internal.build.f_cpu=8000000L

#make atmega328 AVR_FREQ=1000000L BAUD_RATE=9600 UART=0 LED=B5 LED_START_FLASHES=2
328.menu.clock.1MHz_internal=1MHz internal
328.menu.clock.1MHz_internal.upload.speed=9600
328.menu.clock.1MHz_internal.bootloader.low_fuses=0x62
328.menu.clock.1MHz_internal.bootloader.high_fuses=0xd6
328.menu.clock.1MHz_internal.bootloader.file=optiboot_flash/atmega328/optiboot_flash_atmega328_9600_1MHz.hex
328.menu.clock.1MHz_internal.build.f_cpu=1000000L

############################################################
```

sandeepmistry, 2016-07-08:
Hi @MCUdude,

Have you tried something like this?

```
328.menu.clock.16MHz_external.bootloader.file=optiboot_flash/atmega328/optiboot_flash_{build.mcu}_115200_16MHz.hex

```

Since you have have the following lines, `build.mcu` should already be set:
```
328.menu.variant.modelP=328P / 328PA
328.menu.variant.modelP.build.mcu=atmega328p
328.menu.variant.modelNone=328 / 328A
328.menu.variant.modelNone.build.mcu=atmega328
```

MCUdude, 2016-07-08:
Great, thank you so much! 😃 The information about creating a feature rich boards.txt file is a bit short, so I'll leave my file here just for reference.

Actually I took it one step further and made it select the baud rate by itself. I've tested it using an ATmega328 and ATmega328p, and it works perfectly!
`328.menu.clock.16MHz_external.bootloader.file=optiboot_flash/{build.mcu}/optiboot_flash_{build.mcu}_{upload.speed}_16MHz.hex
`

```
# MiniCore - An Arduino core for the ATmega48/P, ATmega8, ATmega88/P, ATmega168/P, ATmega328/P
# https://github.com/MCUdude/MiniCore

menu.clock=Clock
menu.BOD=B.O.D
menu.variant=Variant
############################################################

328.name=ATmega328
328.upload.tool=arduino:avrdude
328.upload.protocol=arduino
328.upload.maximum_size=32256
328.upload.maximum_data_size=2048

328.bootloader.tool=arduino:avrdude
328.bootloader.unlock_bits=0x3f
328.bootloader.lock_bits=0x0f

328.build.core=arduino:arduino
328.build.board=AVR_ATmega328
328.build.variant=standard

328.menu.variant.modelP=328P / 328PA
328.menu.variant.modelP.build.mcu=atmega328p
328.menu.variant.modelNoneP=328 / 328A
328.menu.variant.modelNoneP.build.mcu=atmega328

328.menu.BOD.2v7=2.7v
328.menu.BOD.2v7.bootloader.extended_fuses=0xfd
328.menu.BOD.4v3=4.3v
328.menu.BOD.4v3.bootloader.extended_fuses=0xfc
328.menu.BOD.1v8=1.8v
328.menu.BOD.1v8.bootloader.extended_fuses=0xfe
328.menu.BOD.disabled=Disabled
328.menu.BOD.disabled.bootloader.extended_fuses=0xff

#make atmega328 AVR_FREQ=16000000L BAUD_RATE=115200 UART=0 LED=B5 LED_START_FLASHES=2
328.menu.clock.16MHz_external=16MHz external
328.menu.clock.16MHz_external.upload.speed=115200
328.menu.clock.16MHz_external.bootloader.low_fuses=0xf7
328.menu.clock.16MHz_external.bootloader.high_fuses=0xd6
328.menu.clock.16MHz_external.bootloader.file=optiboot_flash/{build.mcu}/optiboot_flash_{build.mcu}_{upload.speed}_16MHz.hex
328.menu.clock.16MHz_external.build.f_cpu=16000000L

#make atmega328 AVR_FREQ=20000000L BAUD_RATE=115200 UART=0 LED=B5 LED_START_FLASHES=2
328.menu.clock.20MHz_external=20MHz external
328.menu.clock.20MHz_external.upload.speed=115200
328.menu.clock.20MHz_external.bootloader.low_fuses=0xf7
328.menu.clock.20MHz_external.bootloader.high_fuses=0xd6
328.menu.clock.20MHz_external.bootloader.file=optiboot_flash/{build.mcu}/optiboot_flash_{build.mcu}_{upload.speed}_20MHz.hex
328.menu.clock.20MHz_external.build.f_cpu=20000000L

#make atmega328 AVR_FREQ=12000000L BAUD_RATE=57600 UART=0 LED=B5 LED_START_FLASHES=2 
328.menu.clock.12MHz_external=12MHz external
328.menu.clock.12MHz_external.upload.speed=57600
328.menu.clock.12MHz_external.bootloader.low_fuses=0xf7
328.menu.clock.12MHz_external.bootloader.high_fuses=0xd6
328.menu.clock.12MHz_external.bootloader.file=optiboot_flash/{build.mcu}/optiboot_flash_{build.mcu}_{upload.speed}_12MHz.hex
328.menu.clock.12MHz_external.build.f_cpu=12000000L

#make atmega328 AVR_FREQ=8000000L BAUD_RATE=38400 UART=0 LED=B5 LED_START_FLASHES=2 
328.menu.clock.8MHz_external=8MHz external
328.menu.clock.8MHz_external.upload.speed=38400
328.menu.clock.8MHz_external.bootloader.low_fuses=0xf7
328.menu.clock.8MHz_external.bootloader.high_fuses=0xd6
328.menu.clock.8MHz_external.bootloader.file=optiboot_flash/{build.mcu}/optiboot_flash_{build.mcu}_{upload.speed}_8MHz.hex
328.menu.clock.8MHz_external.build.f_cpu=8000000L

#make atmega328 AVR_FREQ=8000000L BAUD_RATE=38400 UART=0 LED=B5 LED_START_FLASHES=2 
328.menu.clock.8MHz_internal=8MHz internal
328.menu.clock.8MHz_internal.upload.speed=38400
328.menu.clock.8MHz_internal.bootloader.low_fuses=0xe2
328.menu.clock.8MHz_internal.bootloader.high_fuses=0xd6
328.menu.clock.8MHz_internal.bootloader.file=optiboot_flash/{build.mcu}/optiboot_flash_{build.mcu}_{upload.speed}_8MHz.hex
328.menu.clock.8MHz_internal.build.f_cpu=8000000L

#make atmega328 AVR_FREQ=1000000L BAUD_RATE=9600 UART=0 LED=B5 LED_START_FLASHES=2
328.menu.clock.1MHz_internal=1MHz internal
328.menu.clock.1MHz_internal.upload.speed=9600
328.menu.clock.1MHz_internal.bootloader.low_fuses=0x62
328.menu.clock.1MHz_internal.bootloader.high_fuses=0xd6
328.menu.clock.1MHz_internal.bootloader.file=optiboot_flash/{build.mcu}/optiboot_flash_{build.mcu}_{upload.speed}_1MHz.hex
328.menu.clock.1MHz_internal.build.f_cpu=1000000L

############################################################
```

sandeepmistry, 2016-07-08:
Very cool!

Feel free to edit the wiki page with any tips or examples:

https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification#custom-board-menus

closed 2016-07-08 by sandeepmistry

MCUdude, 2016-07-08:
Oh, I got one more boards.txt related question; how can I rearrange the order of the sub menus?
I'd like the order to be "variants", "clock" and "BOD".

[image]

sandeepmistry, 2016-07-08:
On first glance, it doesn't look like you can change the order of the entries now. Everything is stored in a map, so the order is alphabetical.