arduino/Arduino #5109: Arduino IDE not start on MAC OS X

holdenpies, 2016-07-08:
Hi,
I have been trying to install the arduino IDE on my mac.
I tried in IDE version 1.6.9 and the IDE does not load at all.
I updated the java and mac software versions and tried again to no success (now mac version 10.11.5
java 8.91)
I tried running in 32 bit mode from 'file info' again no success
I tried a few other versions (1.6.8, 1.0.9 and the recent hourly build) again with no real success

After more searching I found version 1.6.1 (https://www.arduino.cc/en/Guide/IDEonRecentMacOSX) which seemed to be built address the problem.  This is the most successful version so far, the IDE loads.  When I try and upload a sketch however i get the following.
"Cannot run program "/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++": error=13, Permission denied"
As requested in the 1.6.1 page I have raised this issue.
Could you give me some advice on how to resolve please?

Thanks

facchinm, 2016-07-11:
Hi @holdenpies ,
could you try to launch the latest IDE in commandline mode?
To do so, open a `Terminal`, write `/Applications/Arduino.app/Contents/MacOS/Arduino` and press enter, then paste here the resulting error.
Thanks!

holdenpies, 2016-07-11:
Hi,

I did as instructed on version 1.6.9.  The feedback was.

"java.io.IOException: Cannot run program "uname" (in directory "."):
error=0, spawn failed

at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)

at java.lang.Runtime.exec(Runtime.java:620)

at
org.apache.commons.exec.launcher.Java13CommandLauncher.exec(Java13CommandLauncher.java:58)

at org.apache.commons.exec.DefaultExecutor.launch(DefaultExecutor.java:254)

at
org.apache.commons.exec.DefaultExecutor.executeInternal(DefaultExecutor.java:319)

at org.apache.commons.exec.DefaultExecutor.execute(DefaultExecutor.java:160)

at org.apache.commons.exec.DefaultExecutor.execute(DefaultExecutor.java:147)

at processing.app.macosx.Platform.discoverRealOsArch(Platform.java:74)

at processing.app.macosx.Platform.init(Platform.java:66)

at processing.app.Base.guardedMain(Base.java:153)

at processing.app.Base.main(Base.java:135)

Caused by: java.io.IOException: error=0, spawn failed

at java.lang.UNIXProcess.forkAndExec(Native Method)

at java.lang.UNIXProcess.<init>(UNIXProcess.java:248)

at java.lang.ProcessImpl.start(ProcessImpl.java:134)

at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)

... 10 more"

Thanks

facchinm, 2016-07-11:
It looks like your system is unable to spawn any process ( `uname` is used to get the system version). Do you have enforced protections and/or an aggressive Gatekeeper/antivirus?

holdenpies, 2016-07-12:
Hi,

I don't have an aggressive antivirus.   With regard to 'enforced protections' is this System Integrity Protection?  

Thanks

facchinm, 2016-07-12:
Maybe [this](http://stackoverflow.com/a/31015371) post can help, since it reports the same issue you are experiencing, but since the IDE bundles the Java runtime I really can't see how some file could lose executable permissions...

manpits, 2017-10-22:
Just unzip app to /Application
Then execute command "chmod -R go+x /Applications/Arduino.app/Contents/" from terminal
And run app by execure command "sudo /Applications/Arduino.app/Contents/MacOS/Arduino"

xenon129, 2019-02-17:
There is an easier way to make it work under these circumstances (permission problem):
1. Delete the app from /Applications
2. Unzip again using root privileges, so that the unzipped app has proper permissions:

Open terminal app, run these commands:

cd /Applications
sudo unzip /Users/MYUSERNAME/Downloads/arduino-1.8.8-macosx.zip 

enter your user password (assuming you have admin privileges on mac) when sudo prompts you

3. After this, can open the app normally, no need to run sudo every time