arduino/Arduino #5115: Are built-in examples on GitHub too? I'd like to propose an improvement to the Fade sketch

YakovL, 2016-07-10:
Hello Arduino Dev community, I've searched this repo but haven't found, so please guide me where the built-in examples of Arduino IDE are stored and can I propose some improvements?

More specificly, the [Fade](https://www.arduino.cc/en/Tutorial/Fade) sketch has these lines:

    if (brightness == 0 || brightness == 255) {
      fadeAmount = -fadeAmount ;
    }

which I'd say is not a good code as if we change `fadeAmount` to, say, `7` than we will miss the `== 255` value (`brightness` will rise up to 7*36 = `252` and then will become `259` and will grow further). It should be

    if (brightness <= 0 || brightness >= 255) {
      fadeAmount = -fadeAmount ;
    }

instead, I would say. So, where do I propose such improvements?

PaulStoffregen, 2016-07-10:
https://github.com/arduino/Arduino/blob/master/build/shared/examples/01.Basics/Fade/Fade.ino

matthijskooijman, 2016-07-10:
I noticed the pullrequest, so seems this one is answered. Closing.

closed 2016-07-10 by matthijskooijman