arduino/Arduino #5121: RFE: IDE to save board type inside the project

bcraigie, 2016-07-12:
This is a request for enhancement.  I hope you will consider it...

I am developing code for multiple types of Arduino and sometimes I work on two different Arduinos at the same time (in two Arduino IDE windows on the one PC).  When I set the board type to, say, Due in one project and Nano in the other project, when I save them and load them later, the IDE does not remember which board was used for each project and just selects the most recently used board type.  It is essential for many projects to use the correct choice of board architecture for each project.

So my suggestion is to enable the user to choose to remember the board type and save this choice inside the project file.  So that when we open a project that has been saved in this way, the IDE automatically sets the correct board type.  I envisage this could be done by having a Tick box [check box]  for "Remember board type when saving project" in the preferences, or something the user can tick (or untick) before saving (to add or to remove the board type from the saved .ino file).

I appreciate that many of the examples will work with any board type, so I am not suggesting to change the format of these examples or any existing .ino files - only to add an extra section when a user saves their project if they choose to do so.  This extra section, if it exists, will set the board type, and if it doesn't exist, the IDE does what it currently does and uses the board type that was last used.

Thanks for considering this.

Brian

sleemanj, 2016-07-20:
Addition: it should also (be able to) save the selections from the Tools menu.  Many cores allow the user to select from various options in the Tools menu after they have chosen the "Board".  Currently the only way to indicate to the user who is building the sketch which options they should choose is to write them in a comment and tell the user to manually select them.

For example, from [one of my ATtiny13 Examples](https://github.com/sleemanj/ATTinyCore/blob/master/avr/libraries/ATTinyCore/examples/Tiny13/04.Communication/SerialDigitalClock/SerialDigitalClock.ino) I have to have this in a comment...

      Recommended Settings For This Sketch
      ------------------------------------------------------------------------------
      (* indicates non default)
    
      Tools > Board                 : ATTiny13
      Tools > Processor Version     : ATTiny13
      Tools > Use Bootloader        : No (ISP Programmer Upload)
      Tools > Processor Speed       : 9.6MHz Internal Oscillator
      Tools > Millis, Tone Support  : Millis Available, No Tone
    * Tools > Millis Accuracy       : 10%
    * Tools > Print Support         : Dec Only Supported
      Tools > Serial Support        : Half Duplex, Read+Write
     
Which is not exactly kind to the user.

bcraigie, 2016-07-20:
Yes.

Also I notice that with the current IDEs as of today, if you have two projects open, set one to, say Due and the other to, say Nano, then go back to the Due one, it has changed back to Nano, so it looks like the board settings apply globally to all currently open projects, rather than changing only the currently selected project.  So that would need to change as a pre-requisite to these other changes suggested so far.

bbbnh, 2016-08-25:
I don't have a lot to add to this but it seems like it hasn't gotten much attention, so I guess I'll put in my  2c: I was initially more concerned with the same issue as @bcraigie, but I think that and the original are of equal importance and would be huge QoL improvements for developers with multiple types of devices. I'm mostly curious why the decision was made to have the board selection linked between all windows, especially because of the presence of tabs as a way of grouping files that are part of the same sketch; it seems like this is either a "feature" from before tabs were implemented (if there is such a time) or an artifact of some sort of misguided memory management scheme.

per1234, 2017-03-21:
Duplicate of #1481

closed 2017-07-01 by per1234

jdliue, 2019-12-30:
There's a good workaround about this issue. https://forum.arduino.cc/index.php?topic=412121.0