arduino/Arduino #5124: IDE 1.6.9 + ArduinoISP gets "Could not find USBtiny device (0x2341/0x49)" error message.

HP-Sparks, 2016-07-12:
Hi,
I am trying to program an ATtiny85 from Arduino IDE 1.6.9, under Linux (Ubuntu -based machine) after following the indications from there:
- https://github.com/damellis/attiny
 - - http://highlowtech.org/?p=1695
 - - http://highlowtech.org/?p=1706

The ArduinoISP seems fine: "HeartBeat" LED is glowing non stop.
Trying to push the "blink" code to the ATtiny85 ends with this exact error message:
`avrdude: Error: Could not find USBtiny device (0x2341/0x49)` while all necessary settings from the menu are set properly (i.e. Board: ATtiny25/45/85 ; Processor: ATtiny85 ; Clock: Internal 1MHz ; Port: /dev/ttyUSB0 ; Porgrammer: **ArduinoISP**)

Port is /dev/ttyUSB since using an Arduino clone (WaveShare "UNO Plus") with a switch to disable the reset from DTR signal.

I also tried with an Arduino UNO R3 original, same issue: Whatever programmer I select, there is an automatic selection for USBtiny over-riding my selection.
I faced the same issue prior to engage into ATtiny85 development.

Any idea for a workaround ?

Thanks in advance !

Best regards,
  HP_

HP-Sparks, 2016-07-14:
Hi,
I found the origin of my problem: "ArduinoISP" versus "Arduino as ISP" for the programmer choice.
Curiously, I had Not the "Arduino as ISP" choice when I tried yesterday.
Changes since yesterday are:
 - Forced reboot of my system (full swap and memory issue...), probably explaining other issues with sub-menus displayed shifted, time to time...
 - Downgrade of the version of the David A. Mellis (https://raw.githubusercontent.com/damellis/attiny/ide-1.6.x-boards-manager/package_damellis_attiny_index.json) "cores" definitions, from v1.0.2 to v1.0.1
 - Then, restatblished the latest "cores" version at v1.0.2

Sorry for this false issue.

Best regards,
  HP_

closed 2016-07-14 by HP-Sparks

agdl, 2016-07-14:
@HP-Sparks thank you for reporting this. However I suggest you to use this core for tiny devices:
https://github.com/SpenceKonde/ATTinyCore

dnloliveira1, 2016-10-22:
Aconteceu comigo a mesma coisa, obrigado pelo apoio kkkkkkkk

positron96, 2021-09-20:
Am I right that "ArduinoISP" is broken and does not work with ArduinoISP sketch? 
My Arduino 1.8.13 has 3 entries of bootloaders: "ArduinoISP", "ArduinoISP.org" and "Arduino as ISP", of which only the latter works. First 2 complain about "avrdude: Error: Could not find USBtiny device (0x2341/0x49)"