arduino/Arduino #5131: Arduino requires X11 DISPLAY where no should be needed

dagwieers, 2016-07-13:
If you use Arduino from the command-line, it complains about requiring an X11 DISPLAY:

```
$ arduino --pref boardsmanager.additional.urls=http://arduino.esp8266.com/stable/package_esp8266com_index.json --save-prefs
Picked up JAVA_TOOL_OPTIONS: 
Exception in thread "main" java.lang.ExceptionInInitializerError
Caused by: java.awt.HeadlessException: 
No X11 DISPLAY variable was set, but this program performed an operation which requires it.
	at java.awt.SplashScreen.getSplashScreen(SplashScreen.java:117)
	at processing.app.Base.<clinit>(Base.java:91)
```

This forces people to implement ugly work-arounds using Xvfb in order to accomodate Arduino IDE: https://github.com/timelab/ADEM/blob/master/.travis.yml#L5
```yaml
  # Ensure we have a DISPLAY for the Arduino IDE
  - "/sbin/start-stop-daemon --start --quiet --pidfile /tmp/custom_xvfb_1.pid --make-pidfile --background --exec /usr/bin/Xvfb -- :1 -ac -screen 0 1280x1024x16"
  - sleep 3
  - export DISPLAY=:1.0
```

dagwieers, 2016-07-14:
Duplicate of #1981

facchinm, 2016-07-19:
Indeed, closing as duplicate of #1981 (however, the related PR is really good :smile: )

PR #5132 (2016-08-22): Do not show splash screen when options are present
This ensures now splash-screen is shown when preferences are changed or CLI-mode is used.

More information in #5131.

However this does NOT fix #5131, since the Arduino IDE in CLI-mode still requires X11 (according to #1981).
commit 1b95b439 (2016-08-22): Do not show splash screen when options are present

This ensures now splash-screen is shown when preferences are changed or CLI-mode is used.

More information in #5131.

However this does NOT fix #5131, since the Arduino IDE in CLI-mode still requires X11 (according to #1981).
closed 2016-07-19 by facchinm