arduino/Arduino #5136: Wrong type in function puthex() of file ATmegaBOOT_168.c

MichaelGuth, 2016-07-15:
The function void puthex(char ch) of file
arduino-1.6.9/hardware/arduino/avr/bootloaders/atmega/ATmegaBOOT_168.c is wrong. The correct version is:
void puthex(unsigned char ch) {
	unsigned char ah;

	ah = ch >> 4;
	if(ah >= 0x0a) {
		ah = ah - 0x0a + 'a';
	} else {
		ah += '0';
	}
	
	ch &= 0x0f;
	if(ch >= 0x0a) {
		ch = ch - 0x0a + 'a';
	} else {
		ch += '0';
	}
	
	putch(ah);
	putch(ch);
}
The type of ch and ah has to be unsigned char instead of char.

matthijskooijman, 2016-07-15:
For reference, here's the source: https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/bootloaders/atmega/ATmegaBOOT_168.c#L901-L920

Is this actually causing any observable problems? I agree that using unsigned chars would be better here, but I suspect that the end result might be the same. The possibly problematic operation is the bit shift (`ch >> 4`), which apparently will do an "arithmetic shift" when used on signed values, meaning the topmost bit is replicated. I think this means that a "negative" (i.e. >= 128) value will remain negative after shifting, so the `>= 0x0a` will always be false.

However, I'm not sure if this bootloader is actually still used in produced devices, quite possibly they have all been changed to optiboot by now?

MichaelGuth, 2016-07-15:
I used it to print the signate bytes:

#define SIG1    0x1E
#define SIG2    0x95
#define SIG3    0x0F

           puthex(SIG1);
           putch(' ');
           puthex(SIG2);
           putch(' ');
           puthex(SIG3);

"unsigned char" version :  1e 95 0f
"char" version                :  1e )5 0f

-- 
Michael Guth
senior engineering manager

ATecoM realizing visions GmbH
Praemienstr. 29
D-52076 Aachen
Germany

Phone: +49-2408-93802-14
Fax:   +49-2408-93802-20
Email: guth@atecom.com
URL: http://www.atecom.com

Amtsgericht Aachen, HRB# 5941, Sitz: Aachen
Geschäftsführer: Robert Bonnie
USt.-Id. Nr. / VATID: DE 811 66 99 76
Steuernummer/Tax-ID: 225/5775/0558

oqibidipo, 2016-07-17:
> The possibly problematic operation is the bit shift (ch >> 4), which apparently will do an "arithmetic shift" when used on signed values, meaning the topmost bit is replicated. I think this means [...]

The result of right-shifting negative values in C/C++ is implementation-defined. GCC uses sign extension.

MichaelGuth, 2016-07-18:
I try to print 0x95 in my example. The arithmetic shift will result in
0xF9. This is not >= 0x0a and the next calculation is:

ah += '0';  which means
0xF9 + 0x30 = 0x129
0x29 is ASCII character ')' and that is what I get on the terminal output.

So one solution would be to add :
ah &= 0x0f;
between line 904 and 905 if you do not want to change the type of ah and ch.

By the way I am using the avr-gcc in
arduino-1.6.9/hardware/tools/avr/bin

avr-gcc -v outputs :

Using built-in specs.
COLLECT_GCC=avr-gcc
COLLECT_LTO_WRAPPER=/home/guth/253/ATMEL/arduino-1.6.9/hardware/tools/avr/bin/../libexec/gcc/avr/4.8.1/lto-wrapper
Target: avr
Configured with: ../gcc-4.8.1/configure --enable-fixed-point 
--enable-languages=c,c++ 
--prefix=/home/jenkins/jenkins/jobs/toolchain-avr-linux64/workspace/objdir 
--enable-long-long --disable-nls --disable-checking --disable-libssp 
--disable-libada --disable-shared --enable-lto --with-avrlibc=yes 
--with-dwarf2 --disable-doc --target=avr
Thread model: single
gcc version 4.8.1 (GCC)

-- 
Michael Guth
senior engineering manager

ATecoM realizing visions GmbH
Praemienstr. 29
D-52076 Aachen
Germany

Phone: +49-2408-93802-14
Fax:   +49-2408-93802-20
Email: guth@atecom.com
URL: http://www.atecom.com

Amtsgericht Aachen, HRB# 5941, Sitz: Aachen
Geschäftsführer: Robert Bonnie
USt.-Id. Nr. / VATID: DE 811 66 99 76
Steuernummer/Tax-ID: 225/5775/0558