arduino/Arduino #5143: Repeatable builds and snapshots of package/library indexes

binarin, 2016-07-19:
Build process downloads some files from the internet. Most of them are versioned, but following one don't contain any version info in URL and change quite often:
- http://downloads.arduino.cc/packages/package_index.json.gz
- http://downloads.arduino.cc/packages/package_index.json.sig
- http://downloads.arduino.cc/libraries/library_index.json.gz

So it's not possible to package it that way for any system that aims for repeatable builds. E.g. for my NixOS effort (https://github.com/NixOS/nixpkgs/pull/17060) I had to create my own snapshots of those files. 

Could you please provide some sort of daily or weekly snapshots? So any packager can consume them directly from some urls like `http://downloads.arduino.cc/libraries/2016-07-18/library_index.json.gz`

closed 2016-08-12 by cmaglie

cmaglie, 2016-08-12:
Hi @binarin 

those files are no more downloaded to create the build:

- the downloaded `package_index.json*` files have been replaced by a `package_index_bundled.json` that describes only the bundled AVR core and is part of the git tree:
https://github.com/arduino/Arduino/commit/e80c0859963f5f9d695743f4b7dc2d77ec18d4f7

- the `library_index.json` is no more downloaded, since it's not needed to run the IDE (unless you want to install a new library but the Library Manager will take care to download an updated copy anyway in that case)
https://github.com/arduino/Arduino/commit/d5dc479e6b2716244a0e661c1e67ef695401defa

This should solve the reproducible builds problem. Feel free to reopen the issue if I'm missing something.

binarin, 2016-08-12:
Thanks a lot!

commit d5dc479e (2016-08-12): library_index.json is no more bundled.

There is no reason to bundle this file.

If the index file is not available an empty index is
returned by the parser.

Fix #5143
(together with e80c08: Use a specific hardware/package_index_bundled.json)