arduino/Arduino #5144: [Library Manager]: Add gadgetron-software-libraries to the library manager

Guardado, 2016-07-20:
https://github.com/NVSL/gadgetron-software-libraries

Thank you!

agdl, 2016-07-20:
@Guardado this isn't a library but a collection of libraries. Most of them are already present in the library manager however. I think you should ask to the authors however before distribute them in this way

agdl, 2016-07-28:
I'm closing this for lack of feedback. Reopen it if needed

closed 2016-07-28 by agdl

Guardado, 2016-07-31:
Hi Arturo,

I modified the [github repository](https://github.com/NVSL/gadgetron-software-libraries) to include the license of all the libraries we are using from other authors in a file named LICENSE, this libraries where also modified so that they include more documentation and some other functions that we need. The rest of the sources that are not in the LICENSE file is code that we wrote. 

Second, I moved all the headers to one file header so that it doesn't includes multiple headers and conflict with other libraries. Finally I would like to comment that this repository was created as part of a research project that we are doing with an external tool and we would highly appreciate if you could add this repository to the Arduino Library Manager.

Thank you,
Jorge Garza