arduino/Arduino #5146: Boards manager doesn't correctly do cleanup for esp8266 library upgrade

electronicsguy, 2016-07-21:
Please refer to this: [esp8266 v2.2 to v2.3 upgrade compile error](https://github.com/esp8266/Arduino/issues/2297)

The Boards manager doesn't correctly do cleanup for esp8266 library upgrade, supposably.

facchinm, 2016-07-21:
The bug could be caused by the (temporary) removal of v2.2 from the json (so the board manager doesn't know what he needs to remove and you end up with a copy of the core). However, the whole upgrading thing should be reworked in a more reliable way.
https://github.com/arduino/arduino-builder/pull/84 could make the issue less difficult to debug in the meantime.

ektor5, 2016-08-25:
Hi there,
I have this problem too, and arduino/arduino-builder#84 could be a good solution. Why it isn't applied yet? :)

Ek5