arduino/Arduino #5147: software

HerperProduction, 2016-07-21:
hello my software on my computer doesnt it puts me a message that I forgot the sketchbook Arduino thank you to answer me as soon as possible in French if possible 
thank you 
goodbye

facchinm, 2016-07-22:
Hi @HerperProduction , your problem is probably with the path you chose to install Arduino IDE which overrides the default sketchbook location ($your_home_folder/Arduino). 
Move the IDE somewhere else and you should be able to start it. If it doesn't work, try with the [forum](https://forum.arduino.cc/), there is also a section in [French](https://forum.arduino.cc/index.php?board=33.0)

==== autotranslated version :smile: ======

Salut @HerperProduction , votre problème est probablement avec le chemin que vous avez choisi d'installer Arduino IDE qui remplace l'emplacement de sketchbook par défaut ($ your_home_folder / Arduino ) .
Déplacer l' IDE ailleurs et vous devriez être en mesure de démarrer. Si cela ne fonctionne pas, essayez avec le [ forum ] ( https://forum.arduino.cc/ ) , il y a aussi une section en [français ] ( https://forum.arduino.cc/index.php ? pension = 33,0 )

closed 2016-07-22 by facchinm

HerperProduction, 2016-07-22:
Merci pour votre réponse mai que ce que c'est l'IDE
Merci
@HerperProduction