arduino/Arduino #5148: VirtulaWire and Servo libraries

pararoglu, 2016-07-21:
libraries\Servo\avr\Servo.cpp.o: In function `__vector_11':

C:\Users\....\Documents\Arduino\libraries\Servo\src\avr/Servo.cpp:81: multiple definition of `__vector_11'

libraries\VirtualWire\VirtualWire.cpp.o:C:\Users\...\Documents\Arduino\libraries\VirtualWire/VirtualWire.cpp:591: first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Uno.

Here is the Error message. This message appears when using servo and virtualwire library  in same time.

facchinm, 2016-07-22:
Hi @pararoglu ,
both VirtualWire and Servo use TIMER1 interrupts capabilities. Since the Uno only has three timers and timer1 is the only 16bit one it's the only reasonable choice. 
Do you really need to use VirtualWire instead than Wire? 
Anyway, try posting on the [forum](http://forum.arduino.cc/index.php?board=9.0) to get alternatives/help with your project!
Thanks

closed 2016-07-22 by facchinm

pararoglu, 2016-07-22:
I know that both use Timer2. Do you offer another library for RF communication?

facchinm, 2016-07-22:
Servo uses timer1 (https://github.com/arduino-libraries/Servo/blob/master/src/avr/ServoTimers.h#L56) and VirtualWire too (https://github.com/danielesteban/ArduinoLib/blob/master/VirtualWire/VirtualWire.cpp#L566). Which RF chip are you targeting? Connecting it to SCL and SDA pins lets you use plain Wire and get rid of the conflict

cmaglie, 2016-07-22:
VirtualWire is not a replacement for Wire (nor viceversa), besides the similarity in the name.

VirtualWire implements a binary protocol to transfer data instead Wire is an I²C client.

facchinm, 2016-07-22:
This explains a lot, thanks @cmaglie :wink: