arduino/Arduino #5152: Quick question regarding 2.1 library specification.

Chris--A, 2016-07-24:
Is the new library specification (rev 2.1) backwards compatible.

As in the 'includes' property isn't going to cause issues?

cmaglie, 2016-08-05:
> Is the new library specification (rev 2.1) backwards compatible.

yes the `includes` property is optional.

> Also, when testing with 1.5.3, I get this error on many libraries:
> > Missing 'core-dependencies' from library

IIRC this enforcing was present in 1.5.3 and removed afterward. Maybe the specification is not fully precise in this matter. Have you tried with later releases like 1.5.4 or 1.5.5?

cmaglie, 2016-08-18:
I'm closing this one, feel free to reopen if you have further questions.

closed 2016-08-18 by cmaglie