arduino/Arduino #5154: Crash on OSX El Capitan

PhoenixFnX, 2016-07-24:
Here is an error I get at start: 

Exception non détectée dans la méthode principale : java.lang.NoClassDefFoundError: org/fife/ui/rsyntaxtextarea/RSyntaxTextArea

[https://snag.gy/v07hXT.jpg](https://snag.gy/v07hXT.jpg
)

```

nicolasthery@imac ~ $ /Applications/Arduino.app/Contents/MacOS/JavaApplicationStub 
[LaunchRunner Error] processing.app.Base.main(String[]) threw an exception:
java.lang.NoClassDefFoundError: org/fife/ui/rsyntaxtextarea/RSyntaxTextArea
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClassCond(ClassLoader.java:637)
	at java.lang.ClassLoader.defineClass(ClassLoader.java:621)
	at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:141)
	at java.net.URLClassLoader.defineClass(URLClassLoader.java:283)
	at java.net.URLClassLoader.access$000(URLClassLoader.java:58)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:197)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:190)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:306)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:301)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:247)
	at processing.app.Base.handleOpen(Base.java:765)
	at processing.app.Base.restoreSketches(Base.java:407)
	at processing.app.Base.<init>(Base.java:322)
	at processing.app.Base.main(Base.java:167)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:39)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:25)
	at java.lang.reflect.Method.invoke(Method.java:597)
	at apple.launcher.LaunchRunner.run(LaunchRunner.java:116)
	at apple.launcher.LaunchRunner.callMain(LaunchRunner.java:51)
	at apple.launcher.JavaApplicationLauncher.launch(JavaApplicationLauncher.java:52)
Caused by: java.lang.ClassNotFoundException: org.fife.ui.rsyntaxtextarea.RSyntaxTextArea
	at java.net.URLClassLoader$1.run(URLClassLoader.java:202)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:190)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:306)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:301)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:247)
	... 23 more
Exception in thread "AWT-EventQueue-0" java.lang.NoClassDefFoundError: org/fife/ui/rsyntaxtextarea/RSyntaxTextArea
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClassCond(ClassLoader.java:637)
	at java.lang.ClassLoader.defineClass(ClassLoader.java:621)
	at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:141)
	at java.net.URLClassLoader.defineClass(URLClassLoader.java:283)
	at java.net.URLClassLoader.access$000(URLClassLoader.java:58)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:197)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:190)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:306)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:301)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:247)
	at processing.app.Base.handleQuit(Base.java:901)
	at processing.app.macosx.ThinkDifferent.handleQuit(ThinkDifferent.java:124)
	at com.apple.eawt._AppEventLegacyHandler$7.dispatchEvent(_AppEventLegacyHandler.java:146)
	at com.apple.eawt._AppEventLegacyHandler.sendEventToEachListenerUntilHandled(_AppEventLegacyHandler.java:168)
	at com.apple.eawt._AppEventLegacyHandler.handleQuitRequestWith(_AppEventLegacyHandler.java:144)
	at com.apple.eawt._AppEventHandler$_QuitDispatcher.performUsing(_AppEventHandler.java:387)
	at com.apple.eawt._AppEventHandler$_QuitDispatcher.performUsing(_AppEventHandler.java:380)
	at com.apple.eawt._AppEventHandler$_AppEventDispatcher$1.run(_AppEventHandler.java:489)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:209)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:715)
	at java.awt.EventQueue.access$400(EventQueue.java:82)
	at java.awt.EventQueue$2.run(EventQueue.java:676)
	at java.awt.EventQueue$2.run(EventQueue.java:674)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.AccessControlContext$1.doIntersectionPrivilege(AccessControlContext.java:86)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:685)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:296)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:211)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:196)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:188)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:122)
Caused by: java.lang.ClassNotFoundException: org.fife.ui.rsyntaxtextarea.RSyntaxTextArea
	at java.net.URLClassLoader$1.run(URLClassLoader.java:202)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:190)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:306)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:301)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:247)
	... 34 more

```

facchinm, 2016-07-25:
Hi, the name of the application you are launching is still `JavaApplicationStub`, which means  that you are using a pre 1.6.2 IDE. Could you try with the latest one (1.6.9) and report the results?

sandeepmistry, 2016-08-24:
I'm closing this due to lack of activity, please re-open with requested info. is you are still interested.

closed 2016-08-24 by sandeepmistry