arduino/Arduino #5156: Serial Libraries imply RS-232 support

DavidHSilber, 2016-07-25:
None of the pages which describe the Serial or SoftwareSerial libraries mention that the resulting communication stream is not actually RS-232. This is something that is likely obvious to people experienced in the hardware side of things, but it was not in the least obvious to me coming from a software background. A simple mention of this fact would have saved me days of effort.

DavidHSilber, 2016-07-25:
Never mind. On re-reading the Serial page I see a mention that RS-232 works at different voltage levels. No idea how I missed that.

DavidHSilber, 2016-07-25:
I see now why I missed this important comment. It is buried at the end of a paragraph discussing the Arduino Mega. Since that was not the hardware I was working with, I skipped those details. Please move the parenthetical comment out of that paragraph and place it in a paragraph of its own.

https://www.arduino.cc/en/Reference/Serial

agdl, 2016-07-25:
Done. Stated as first paragraph. Thank you!

closed 2016-07-25 by agdl