arduino/Arduino #5157: [Library Manager] New library BowlerCom for communication and robotics

madhephaestus, 2016-07-25:
Hello,

I have ported over my robotics communication and control framework to an Arduino 1.5 compliant stack. 

It can be found at:

https://github.com/CommonWealthRobotics/BowlerCom

Thank you!

agdl, 2016-07-25:
@madhephaestus please fix library.properties. Look here
https://github.com/CommonWealthRobotics/BowlerCom/blob/3.12.1/library.properties

madhephaestus, 2016-07-25:
I'm sorry, I don't understand what's wrong. Did a log file kick back any information about whats wrong?

madhephaestus, 2016-07-25:
Hows this, i made a new release: https://github.com/CommonWealthRobotics/BowlerCom/blob/0cf0d1f9829ab5dc96f90e83cfe5a4665e3b4068/library.properties

madhephaestus, 2016-07-25:
Ah, yes, i had a mismatch between the release number in the file and the Git release, that is resolved now in 3.12.2

madhephaestus, 2016-07-25:
@agdl I believe it should be fixed now.

agdl, 2016-07-26:
Done thank you!

closed 2016-07-26 by agdl