arduino/Arduino #5158: Library register request: WizFi250 library

kaizen8501, 2016-07-26:
Hi. I made Arduino library for Wi-Fi.
WizFi250 is Wi-Fi shield and I think this library helps using your Arduino board with Wi-Fi easily and fast.
Please add my library to Library manager.

This is gitbug url : https://github.com/Wiznet/WizFi250_arduino_library

Thank you.

agdl, 2016-07-26:
Hi, please follow specification about repository structure

https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification

kaizen8501, 2016-07-26:
Thank you for fast reply. I modified directory structure.
Please check my repository again ^^
Thank you

agdl, 2016-07-26:
Structure is good, but a tag is missing. Please follow [this guide](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ)

agdl, 2016-07-26:
Another note. About the readme:
is Arduino not Arudino

kaizen8501, 2016-07-26:
Hi. I modified wrong word and made tag to V1.0.
Please check again ^^

agdl, 2016-07-26:
Done thank you!

closed 2016-07-26 by agdl