arduino/Arduino #5159: Arduino port selection greyed out

geek1011, 2016-07-26:
The port selection is greyed out and I cannot upload.

```
$ arduino
Loading configuration...
Exception in thread "cc.arduino.packages.discoverers.serial.SerialBoardsLister" java.lang.NullPointerException
	at processing.app.Platform.resolveDeviceByVendorIdProductId(Platform.java:143)
	at processing.app.linux.Platform.resolveDeviceAttachedTo(Platform.java:142)
	at cc.arduino.packages.discoverers.serial.SerialBoardsLister.run(SerialBoardsLister.java:83)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)
ABRT problem creation: 'success'
Initializing packages...
Preparing boards...
Starting...
$ id
uid=1000(patrick) gid=1000(patrick) groups=1000(patrick),10(wheel),18(dialout),54(lock),978(vboxusers)
$ uname -a
Linux fedora 4.5.7-300.fc24.x86_64 #1 SMP Wed Jun 8 18:12:45 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
$ ls -l /dev/ttyACM0
crw-rw---- 1 root dialout 166, 0 Jul 25 20:43 /dev/ttyACM0
$ dmesg | tail
[  152.019015] usb 3-9: new full-speed USB device number 6 using xhci_hcd
[  152.185226] usb 3-9: New USB device found, idVendor=2341, idProduct=0043
[  152.185229] usb 3-9: New USB device strings: Mfr=1, Product=2, SerialNumber=220
[  152.185230] usb 3-9: Manufacturer: Arduino (www.arduino.cc)
[  152.185231] usb 3-9: SerialNumber: 75533343336351507202
[  152.277863] cdc_acm 3-9:1.0: ttyACM0: USB ACM device
[  152.278110] usbcore: registered new interface driver cdc_acm
[  152.278111] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
```

agdl, 2016-07-26:
@geek1011 Arduino 1.6.5 is an old release. Can you please try our latest hourly builds or latest stable?

https://www.arduino.cc/en/Main/Software

geek1011, 2016-07-26:
I'll try this as soon as I can.

sandeepmistry, 2016-08-24:
Hi @geek1011, have you had a chance to try out the latest IDE release?

geek1011, 2016-08-28:
Yes, sorry for the wait, I forgot to respond. It works now.

closed 2016-08-28 by geek1011

Vinayakj009, 2017-05-03:
Hi. I saw something peculear happening in arduino-1.8.2, on ubuntu 16.04.02 LTS.
If I run ./arduino from the arduino home directory, while it gives the exception
Exception in thread "cc.arduino.packages.discoverers.serial.SerialBoardsLister" java.lang.IndexOutOfBoundsException: Index: 0, Size: 0
at java.util.LinkedList.checkElementIndex(LinkedList.java:555)
at java.util.LinkedList.get(LinkedList.java:476)
at processing.app.Platform.resolveDeviceByVendorIdProductId(Platform.java:241)
at cc.arduino.packages.discoverers.serial.SerialBoardsLister.retriggerDiscovery(SerialBoardsLister.java:113)
at cc.arduino.packages.discoverers.serial.SerialBoardsLister.run(SerialBoardsLister.java:183)
at java.util.TimerThread.mainLoop(Timer.java:555)
at java.util.TimerThread.run(Timer.java:505)

The peculear part is, it gives this exception only once and when the usb device is connected.

Also if I run the command without the device connected, this error does not show up untill I attach the usb device.

I have had this problem before, and sometimes it does not happen. Also when it does, my pc becomes useless for arduino for days togather.

Note:
I tried all the tricks.
Adding user to dialout group,
running arduino as sudo
connecting disconnecting the usb port several times
etc etc