arduino/Arduino #5164: IDE menu language showing in Cyrillic, Only non US-ASCII characters displayed

thewriterben, 2016-07-26:
I'm currently using Windows 10 (64bit) have been unable to use the IDE due to issues with the written language on the dropdown menu is in Russian script. I've tried uninstalling and reinstalling, cleaning the registry between installs, I tried running a previous version to no avail. My language is set at "System Default" and I tried setting it as English, but so far nothing has helped.
[image]
[image]

facchinm, 2016-07-29:
Hi @thewriterben , if the preferences menu is in English it means that at least a part of the gui has been correctly configured. I believe there is some stray configuration file somewhere overriding your choice. To start from scratch, close the IDE, remove `C:\Users\$yourUser\AppData\Local\Arduino15` and `C:\Users\$yourUser\AppData\Roaming\Arduino15`, move the sketchbook (`C:\Users\$yourUser\Arduino`) somewhere else and restart the IDE. This should make it start in "System Default" language.

c141heaven, 2016-10-11:
I have had a similar issue .... the top menu bar is unreadable.  The drop down menu is also unreadable.  I'm on Win 10, 64 bit, when I get to the preferences screen it's in English.   I've tried system default, and English, and manually edited the preferences.txt file.   exited and restarted the dev env each time, all to no avail.   My windows settings are all set to US locale.

[image]