arduino/Arduino #5165: Wire.h Wire.cpp library consistency: AVR vs. SAM

bowzee, 2016-07-26:
( also discussed here https://forum.arduino.cc/index.php?topic=414891 )

AVR: the constructor for TwoWire is:
`TwoWire();`

SAM: this constructor is:
`TwoWire(Twi *twi, void(*begin_cb)(void), void(*end_cb)(void));`

Why are they different? How do you circumvent this when trying to use a generic AVR application on a SAM (e.g. Due)?

facchinm, 2016-07-29:
Hi @bowzee ,
I am concerned about the implementation using inheritance. Is it really needed? All usages in https://github.com/joeyoung/arduino_keypads/blob/master/Keypad_I2C/Keypad_I2C.cpp are the plain library call, so you can change them into Wire.function() (if a single Wire instance is needed) or myWire->function() if a reference to the TWI instance is provided with the constructor. Then you don't need to worry about implementations since the APIs are shared between all Arduino cores :smile:

closed 2016-07-29 by facchinm

bowzee, 2016-07-29:
Hi @facchinm 

I do not want to rewrite Joe Young's libraries. What I am trying to report is the inconsistency, there is no TwoWire::TwoWire() for SAM (Due), so an exact same program written for AVR will not work with Due (it will not get compiled). Like the examples included with these libraries.

I agree that there probably no need for inheritance (in fact it gets confusing), but that is not my code.

facchinm, 2016-07-29:
The inconsistency is expected (eg. you'll never pretend that SERCOM class will be available on AVR since it's a SAMD feature, but the Serial API is the same between the cores). 
Wire is intended to be used with its singletons (Wire, Wire1 etc) so the constructor is only called by the underlying library. 
I'm sorry but I bet you'll need to slightly modify the library (it's only 3 substitution if you only want to support Wire and not Wire1)

bowzee, 2016-07-29:
Hi @facchinm, 

I find these low level issues very complicated... But OK I will try to alter Joe Young's needed library or libraries (after learning about SERCOM, Wire and Wire1).

Thanks!