arduino/Arduino #5166: Provide sha512 sums

NicoHood, 2016-07-27:
If would be nice if the website could also provide sha512sums of the binaries, as we use this for the AUR packages. Its just safe than md5. Thanks :)

facchinm, 2016-07-29:
Hi Nico, historically only md5 has been provided because it's a commodity of all PCs. But the request is valid so I'm adding another file (named arduino-version.sha512sum.txt ) in downloads.arduino.cc root to check with a collision free hash algorithm :wink:

closed 2016-07-29 by facchinm

NicoHood, 2016-07-29:
Thanks. A website link would also be nice, so you dont have to search them. Also noone will know they even exist. Like this:

Checksums: [md5](https://downloads.arduino.cc/arduino-1.6.10.md5sum.txt) [sha512](https://downloads.arduino.cc/arduino-1.6.10.sha512sum.txt)

facchinm, 2016-07-29:
LGTM :smile:

NicoHood, 2016-07-29:
Thanks a lot :)