arduino/Arduino #5169: Incorrect labels on Zero schematic

catphish, 2016-07-28:
I believe there is an error on the schematic published at https://www.arduino.cc/en/uploads/Main/ArduinoZero-schematic.pdf

In the top-right, connector J102 is labelled as follows:
1 - PA11
2 - PA10
3 - PA08
4 - PA09
5 - PA14
6 - PA15
7 - PA20
8 - DGI_GPIO1 (PA21)

However, in reality, the port is connected as follows:
1 - Digital 0 - PA11
2 - Digital 1 - PA10
3 - Digital 2 - PA14
4 - Digital 3 - PA09
5 - Digital 4 - PA08
6 - Digital 5 - PA15
7 - Digital 6 - PA20
8 - Digital 7 - PA21

The correct information is confirmed here in the source code: https://github.com/arduino/ArduinoCore-samd/blob/master/variants/arduino_zero/variant.cpp

agdl, 2016-08-03:
Done thank you!

catphish, 2016-08-13:
@agdl Thanks, however I still see the incorrect labels on the version currently published at https://www.arduino.cc/en/uploads/Main/ArduinoZero-schematic.pdf

agdl, 2016-08-22:
sorry there was a sync error it is ok now :)

closed 2016-08-22 by agdl