arduino/Arduino #5170: [WEB] Please, present Ethernet library as deprecated (here: https://www.arduino.cc/en/Reference/Libraries, here: https://www.arduino.cc/en/Reference/Ethernet, etc)

q2dg, 2016-07-28:
Why? Because none of current official Arduino products has Ethernet conectors (sadly).

Other solution would be offer a new product: an Ethernet shield with TLS integrated chip and, therefore, modify this library accordingly...dreaming is free.

PD: The same issue should be applied to GSM library, Wifi library, TFT library, Due Audio library, etc in Libraries list page but, well, let's go step by step.

per1234, 2016-07-28:
There's nothing deprecated about it. The library is completely functional, compatible with the current version of the IDE, actively developed, and widely used. Just because Arduino no longer manufactures products that use W5100 doesn't mean the library is of no use to Arduino users, you can still attach a W5100 module to an Arduino.

q2dg, 2016-07-28:
Yes, you're right, but let me expose the reasoning from a (new) user point of view. 

You enter in https://www.arduino.cc/en/Reference/Libraries and see:
     "Ethernet - for connecting to the internet using the Arduino Ethernet Shield "
That's should be changed by:
     "Ethernet - for connecting to the internet using the Wiznet W5100 chip-based shields " or something similar.

You enter in https://www.arduino.cc/en/Reference/Ethernet and the same:
     "With the Arduino Ethernet Shield, this library allows an Arduino board to connect to the internet. (...). Arduino communicates with the shield using the SPI bus."
This should be changed by:
    "With Wiznet W5100 chip-based shields, this library allows ... Arduino communicates with these shields using the SPI bus." or something similar.

But beyond these aesthetic changes, my question is about asking the reason of maintenance of this library in current times.  Let me explain me: if the future of this library is to be not tied to any specific product, it should be more generic (adding more support to chips -W5500 in particular- because W5100 is lagging behind little by little: see the existing "forks" of this library have appeared to support it: https://github.com/Wiznet/WIZ_Ethernet_Library or https://github.com/adafruit/Ethernet2 or https://github.com/embeddist/WIZ_Ethernet_Library-IDE1.5.x for instance). Current situation is no sustainable. It's the same case as TFT library...if it were generic, maybe someone used it (as it is used LiquidCrystal library, precisely) but being tied to an specific chip, if this chip is not associated to an official product, it's a dead-end.

<dreaming mode=on>
On the other hand, if Arduino Team had the idea of reviving Ethernet Shield (with a Wiznet chip embedded or another one), this shield must carry a tls chip as Wifi101 shield does. Nowadays not being capable of using Https is a black hole.
</dreaming mode=off>

Thanks for your attention

Chris--A, 2016-07-28:
If anything I'd change the text `Arduino Ethernet Shield` to simply `Ethernet Shield`.

> On the other hand, if Arduino Team had the idea of reviving Ethernet Shield (with a Wiznet chip embedded or another one), this shield must carry a tls chip as Wifi101 shield does. Nowadays not being capable of using Https is a black hole.

The networking shield producers are probably feeling the pinch as the ESP8266 has taken off. They are great standalone, but are a cheap $3 Wifi shield for your Arduinos. Will get worse when the newer ESP32 takes off.

q2dg, 2016-07-28:
> The networking shield producers are probably feeling the pinch as the ESP8266 has taken off. They are great standalone, but are a cheap $3 Wifi shield for your Arduinos. Will get worse when the newer ESP32 takes off.

But hey, that's wifi, not Ethernet. I think it's another league.

per1234, 2017-07-05:
The Libraries and Ethernet Library reference pages have since been reworked to account for the Ethernet Shield 2 and Ethernet2 library.

There is already an issue for adding W5x00 support to the Ethernet Library: https://github.com/arduino-libraries/Ethernet/issues/37.

I'm closing this as solved.

closed 2017-07-05 by per1234