arduino/Arduino #5171: [WEB] Please, remove Yun shield from Arduino/Genuino Products page

q2dg, 2016-07-28:
There's no link to buy it. I asked to support@arduino.cc about that question and they told me that this product isn't available and there is not ETA. I think it's very confusing and frustrating to see a "ghost" product which can't be bought anywhere, so I suggest this action while this situation isn't resolved. Thanks

jantje, 2016-07-28:
And what for the people who have one?

q2dg, 2016-07-28:
Well, if it is a retired product, its page should be go to https://www.arduino.cc/en/Main/Products#retired
If it is a product is expected to be on sale in a future, its rectangle should be painted as "Coming next" instead of yellowish-orange. It's easy.

jantje, 2016-07-29:
O I missed the web part. Sorry

akash73, 2016-11-07:
Back on product line
https://www.arduino.cc/en/Main/Products

closed 2016-11-07 by akash73