arduino/Arduino #5172: 1.6.10 hardware/tools/avr/avr/include/avr/iom32u4.h causes compile error

MrYsLab, 2016-07-28:
The lines containing:
```
#define PCMSK1 _SFR_MEM8(0x6C)

#define PCMSK2 _SFR_MEM8(0x6D)
```
were removed and this causes compilation errors on sketches that successfully compiled with previous versions of this file.

Please reinstate these lines.

matthijskooijman, 2016-07-28:
Hm, I think 1.6.10 uses a newer avr-libc, so this suggests that these were removed upstream. Did you verify that the 32u4 actually has these registers?

MrYsLab, 2016-07-28:
No, I am just basing this on compiling a sketch I have had in existence since 2013 that uses the [oopinchageint[(https://github.com/GreyGnome/ooPinChangeInt) required by the [adaencoder](library) .

I understand your point, and since the changes came from upstream, will probably not get changed back. 

So I guess I will either have to come up with multiple versions of this sketch or provide my own dummy defines so that the existing libraries (not my own) compiles.

matthijskooijman, 2016-07-28:
If it turns out that the 32u4 does indeed not have these registers, make sure to report a bug with the failing libraries as well, since that is where the proper fix should live in that case.

MrYsLab, 2016-07-28:
Ok, I will close this issue.

closed 2016-07-28 by MrYsLab

oqibidipo, 2016-07-28:
> If it turns out that the 32u4 does indeed not have these registers

That is indeed the case. 32u4 has PCINT on only 8 pins and PCMSK0 covers those. Registers 0x6C and 0x6D are marked as reserved.

matthijskooijman, 2016-07-28:
Thanks for checking and reporting back!