arduino/Arduino #5173: Windows 10 64Bit, Arduino 1.6.10 upload issue.

Siliconbitz, 2016-07-28:
If the IDE is launched by clicking on an .ino it does not appear possible to then 'upload' the compiled code to any of the Arduinos that I tested ( uno & mega ) . If however, the IDE is launched on it's own and the .ino is then loaded ... compilation & uploading will then work,

per1234, 2016-07-28:
Just wanted to confirm that multiple users have reported this problem on the forum when starting the IDE by opening an .ino file and the solution of opening the IDE on its own fixed it.

The only error shown when uploading fails is `An error occurred while uploading the sketch` even when verbose output during upload is checked.

http://forum.arduino.cc/index.php?topic=415264
http://forum.arduino.cc/index.php?topic=415269

facchinm, 2016-07-29:
I just tested a clean install of 1.6.10 on both Win10 and WinXP, the upload works on both when double clicking an ino file. The mystery here is why it works when the IDE is started manually. Maybe the file association points to another installation of the IDE?

amdkt7, 2016-07-30:
I am having the same issue with 1.6.10. I get the "An error occurred while uploading the sketch" if I open the file from explorer, but if I ope the IDE first (the IDE is automatically opening the recent file) it then programs cleanly. I'm using a fresh install of W10 and clean install of 1.6.10 (no previous versions on this machine). Don't know what I can add, just confirming the issue.

Krosik, 2016-07-31:
I think I found it!

For anyone who have error when starting IDE via file and it works when starting IDE via desktop shortcut, open properties of this shortcut and delete "Start in", you will see the error.

Error occurs, because when opening IDE via file, there is no "start in folder" property. Problem will be that IDE is not detecting folder from where it was started, or some functions related to opening COM port do not use path where is IDE stored and try open some files with relative path. Of course when "path" is C:\windows they will fail.

For you facchinm I think you have some path in system's PATH variable, which we do not have.

meoudon, 2016-08-01:
I have solved. It Avrdude's error. I copy worked avrdude from another version into /avr/bin and avrdude.conf. Everything done on 1.6.12.

cmaglie, 2016-08-05:
Is this happening only with 1.6.10 or with 1.6.9 too?

per1234, 2016-08-05:
> Is this happening only with 1.6.10 or with 1.6.9 too?

@cmaglie There are ~15(it's possible that some of the reports here are the same users from the forum, I know Krosik is so I didn't double count him) reports of this issue that were solved by opening the sketch via the IDE instead of opening the sketch directly. Of the 15 there are 3 that didn't state which IDE version they're using. Every other report was using Arduino IDE 1.6.10, no reports stated they were using Arduino IDE 1.6.9.

facchinm, 2016-08-05:
Found it at last, `avrdude-6.3-arduino2` had an unwanted dependency on libusb-0.dll . Launching the IDE by clicking the .ino bypasses the inclusion of IDE root in PATH, so the dll doesn't get loaded at runtime. Since that release of avrdude was quite unlucky we should revert to 6.0.1 in upcoming release, so this problem will be solved (and we must keep in mind to check for it before the next release).

Krosik, 2016-08-05:
with hourly build 2016/08/05, i can **not upload** sketch by any way

```
java.io.IOException: Cannot run program "{runtime.tools.avrdude.path}/bin/avrdude": CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:26)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:130)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:209)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.Sketch.upload(Sketch.java:1187)
	at processing.app.Sketch.exportApplet(Sketch.java:1160)
	at processing.app.Sketch.exportApplet(Sketch.java:1132)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2409)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.io.IOException: CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessImpl.create(Native Method)
	at java.lang.ProcessImpl.<init>(ProcessImpl.java:386)
	at java.lang.ProcessImpl.start(ProcessImpl.java:137)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 9 more
An error occurred while uploading the sketch
```

General-Fault, 2016-08-07:
Also see cases #5189 and #5218.

kamura098, 2016-08-08:
Also having this issue on ide1.6.10,  win10 and an uno

lee-lvd, 2016-08-16:
Yes fresh install of 1.6.10
on Windows 7
AMD 64 bit

double click on MySketch.ino file and it fails to upload to my uno board.

Load the IDE, load the same file
it compiles and uploads with no issues.

Lee

cmaglie, 2016-08-18:
This is solved in IDE 1.6.11. Please reopen if the issue persists.

closed 2016-08-18 by cmaglie

cmaglie, 2016-08-18:
Tracking the build issue here: https://github.com/arduino/avrdude-build-script/issues/3

Tes3awy, 2016-08-21:
Make sure your port under **_Tools>Port_** is **_COM3(Arduino/Genuino Uno)_**

durante, 2016-09-05:
Hi, i have been pulling out my hair with this problem. Arduino Idl 1.6.10, I have many boards, Mega, Leonardo, Due, Uno, Sparkfun Photon Red board and Standard SparkFun Red board.

So all function only sometimes and serial monitor does not work.

I have also tried numerous types of cables to test. What i have found is my cable for my external hard drive with 2 USB port "power / data" works every time! 

I believe that during the upload phase the board is undervoltage at some point causing the fault.

Try uploading with an external power supply connected.

maddy2610, 2017-04-05:
I have bought Rasberry pi from <a href="http://www.w11stop.com/manufacturer-raspberrypi">Rasberry Pi in Pakistan</a> and it has USB (and ethernet) which is not working.. i am confuse i can't understand that is my software is too old  or there is a SD card issue NOOBS issue... 
i have stucked in this.. please can anyone can help me out from this....!!!!

elnaz-t, 2018-12-26:
ذ