arduino/Arduino #5176: Arduino 1.6.10 subfolder error

MagoKimbra, 2016-07-28:
I have module in subfolder, until the 1.6.9 version it could easily include files in subfolder
example #include "module / macros.h"
Now with the new 1.6.10 version of error that can not find the file.
Why?
And 'anything changed in the inclusions?

per1234, 2016-07-29:
You now need to move it to the `src` subfolder of your sketch folder and change the code to:
```arduino
#include "src/module/macros.h"
```
This is the source of the change: https://github.com/arduino/arduino-builder/pull/148

matthijskooijman, 2016-07-29:
Exactly, so I'm closing this commit. @cmaglie, this should be made more prominent in the changelog, I think.

MagoKimbra, 2016-07-29:
Ok.. I change name subfolder "module" in "src" and now function!!! Thank's

closed 2016-07-29 by matthijskooijman