arduino/Arduino #5180: Arduino/Genuino 101-Serial Port not detected Arduino 1.6.12 and Inter Curie 1.0.6  OS X El Capitan-

SingingLioness, 2016-07-29:
Hello,

I just got my first arduino. Its an original 101. However, I am unable to run a single sketch since the serial port  doesn't show my arduino. I have tried reinstalling different IDEs twice going back to 1.6.9. Also, I upgraded to El Capitan from Mountain Lion. I am attaching a screenshot. I am clueless since I am a beginner and also I found no resources specifically for this problem related to El capitan and Arduino 101
[image]

facchinm, 2016-07-29:
Hi @SingingLioness , is the board POWER LED on or is it flickering? If you navigate to `Apple Logo > About This Mac > System Report > Hardware > USB` you can check if the board has been recognised in any way (like DFU mode or similar).
If the LED is ok and the USB menu is empty the board is probably unprogrammed. In this case simply try to upload without selecting any port and it *could* work.
For a better chance to get some help you should try to post your findings in the proper section of the forum [http://forum.arduino.cc/index.php?board=103.0](http://forum.arduino.cc/index.php?board=103.0) (also to help other people who may encounter this problem)
At last, if none of these work you should contact your reseller for a replacement

closed 2016-07-29 by facchinm

SingingLioness, 2016-07-29:
Thanks for the reply @facchinm ! I checked and the MAC does show the Genuino 101. I will post the same on the Arduino forum. Thanks again