arduino/Arduino #5181: [Library Manager] Request to add CheapStepper library to library manager

tyhenry, 2016-07-29:
I've created a library specifically for the 28BYJ-48 stepper motor with ULN2003 driver board.  It's a cheap and readily available stepper, and this library allows you to perform very fine-grained moves, with 4096 steps per rotation.  The library supports blocking and non-blocking moves, direct micro-step control, and comes with 3 example sketches:

[https://github.com/tyhenry/CheapStepper](https://github.com/tyhenry/CheapStepper)

Thanks!

agdl, 2016-08-01:
I'm sorry, but your library can't be added since a tag is missing please follow [this guide](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ)

tyhenry, 2016-08-01:
Hi - thanks for the note.
I've now tagged the repo with release v0.2

agdl, 2016-08-01:
Done thank you!

closed 2016-08-01 by agdl