arduino/Arduino #5183: [Library Manager] Request to add LCDMenuLib to library manager

Jomelo, 2016-07-29:
Hi, i create the lib LCDMenuLib:
https://github.com/Jomelo/LCDMenuLib

You can create with this lib menus for serial monitor, lcd and graphic lcds based on the nested set model.
The lib split the functionality in a display part and a backend part. The display part is only for content on display, in the backend part you can check sensors or other things. The menu can be controlled over serial monitor, buttons (digital or analog), encoder, keypad shield.

The lib have at the moment 10 examples and i planed to include more examples.
 
This lib have a very large thread in german form: http://forum.arduino.cc/index.php?topic=73816.0 (133,843 views)

Thanks!

per1234, 2016-07-30:
From https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ:
>Your repo is ok if its root folder contains file library.properties and the rest of the library source code. If your library is stored into a subfolder, your repo it's not ok and we can't include your library in the list.

You need to move everything from the LCDMenuLib subfolder into the repository root folder, update the version in library.properties, and then create a new release/tag for that version.

Jomelo, 2016-07-31:
Hi, its done:
https://github.com/Jomelo/LCDMenuLib

Version 2.1.4

agdl, 2016-08-01:
Done thank you!

closed 2016-08-01 by agdl