arduino/Arduino #5187: [Library Manager] request to add hd44780 library

bperrybap, 2016-07-30:
https://github.com/duinoWitchery/hd44780

An extensible hd44780 LCD library.
It offers features not availed in other lcd libraries, including full auto location and pin mapping detection for i2c i/o expander based backpacks to provide a "it just works" experience for PCF8574 and MCP23008 based backpacks.

agdl, 2016-08-01:
Done thank you!

closed 2016-08-01 by agdl

bperrybap, 2016-08-01:
It isn't showing up yet. Is there an issue or 
how long does it normally take to show up in the Library manager?
The library manger FAQ says the job runs every hour.
My version and tag is 0.5.0-a which is semver compliant but maybe that revision confuses the git job since it isn't just MM.mm.pp numbers?

bperrybap, 2016-08-01:
The hd44780 library shows up now, but there is an issue.
After the library manager installs the library,  it shows up twice in the library manager. 

I tracked down the problem.
It turns out that the library manager will duplicate an entry if the library name in the .properties file does not exactly match the name in the list of known libraries in the library manager that you guys create.

In this case there is a problem with the library name for this library in the library manager.

The name of the library is hd44780 not HD44780 (case does matter)
The library name for the hd44780 library entry in the library manager is wrong.
Because of this, the entry shows up twice after the library installed since the library manager has a name of **HD44780** and the library .properties says the name is **hd44780** - case matters.

The real name of the library is **hd44780**

Can you please go fix/correct the name in the library manager to be **hd44780** rather than **HD44780** so that the library manager will not duplicate the library and the directory name created under the {libraries} directory will also have the proper/correct library name.

agdl, 2016-08-02:
I fixed it. Sorry for the inconvenience

bperrybap, 2016-08-02:
But now both library entries showing up. 
It looks like now the library manager has both HD44780 and hd44780 entries like the hd44780 was added but the HD44780 was not removed.
I'm not familiar with how the known libraries list for the library manager is created and managed but
if I look in the json file {homedir}/.arduino15/library_index.json I see both entries.

What needs to be done to remove the HD44780 entry?

I have included the entries I see in the library_index.json file below:

```
    {
      "name": "HD44780",
      "version": "0.5.0-a",
      "author": "Bill Perry \u003cbperrybap@opensource.billsworld.billandterrie.com\u003e",
      "maintainer": "Bill Perry \u003cbperrybap@opensource.billsworld.billandterrie.com\u003e",
      "sentence": "Extensible hd44780 LCD library.",
      "paragraph": "hd44780 is an extensible LCD library for hd44780 based LCD displays. The API functionality provided by the hd44780 library class, when combined with an hd44780 library i/o subclass, is compatible with the API functionality of the Arduino LiquidCrystal library as well as most of the LCD API 1.0 Specification (some of which is nearly obsolete). The hd44780 API also provides some addtional extensions and all the API functions provided by hd44780 are common across all i/o subclasses. hd44780 currently includes i/o subclasses for Arduino direct pin control, i2c expander backpacks, and LCDs with native i2c interface.",
      "website": "https://github.com/duinoWitchery/hd44780",
      "category": "Display",
      "architectures": [
        "*"
      ],
      "types": [
        "Contributed"
      ],
      "url": "http://downloads.arduino.cc/libraries/duinoWitchery/HD44780-0.5.0-a.zip",
      "archiveFileName": "HD44780-0.5.0-a.zip",
      "size": 78916,
      "checksum": "SHA-256:79ab6046a402bfcab0e263c105e928dd8f1ded13b49a8c2f0075a566c5acd1b6"
    },
```

And then 80 lines lower:
```
    {
      "name": "hd44780",
      "version": "0.5.0-a",
      "author": "Bill Perry \u003cbperrybap@opensource.billsworld.billandterrie.com\u003e",
      "maintainer": "Bill Perry \u003cbperrybap@opensource.billsworld.billandterrie.com\u003e",
      "sentence": "Extensible hd44780 LCD library.",
      "paragraph": "hd44780 is an extensible LCD library for hd44780 based LCD displays. The API functionality provided by the hd44780 library class, when combined with an hd44780 library i/o subclass, is compatible with the API functionality of the Arduino LiquidCrystal library as well as most of the LCD API 1.0 Specification (some of which is nearly obsolete). The hd44780 API also provides some addtional extensions and all the API functions provided by hd44780 are common across all i/o subclasses. hd44780 currently includes i/o subclasses for Arduino direct pin control, i2c expander backpacks, and LCDs with native i2c interface.",
      "website": "https://github.com/duinoWitchery/hd44780",
      "category": "Display",
      "architectures": [
        "*"
      ],
      "types": [
        "Contributed"
      ],
      "url": "http://downloads.arduino.cc/libraries/duinoWitchery/hd44780-0.5.0-a.zip",
      "archiveFileName": "hd44780-0.5.0-a.zip",
      "size": 78916,
      "checksum": "SHA-256:86f4c443682ea2b7733ed7f6fdabe3ce1c057145f74d24445924573d6dfc93f5"
    },
```

bperrybap, 2016-08-02:
I snooped the IDE traffic when the library manager starts up and noticed it is downloading:
http://downloads.arduino.cc/libraries/library_index.json.gz
If I look in that archive file, it contains library_index.json
and if I look in the json file, it contains entries for both HD44780 and hd44780

Can you update the library list so that it no longer contains the HD44780 entry?

bperrybap, 2016-08-08:
The double entries issue still exists. Can this issue be reopened until this is finally fixed/resolved?

cmaglie, 2016-08-08:
Hi @bperrybap 

sorry for the delay, it should be solved now!

bperrybap, 2016-08-08:
It seems fixed. Thank you.