arduino/Arduino #5188: Error downloading http://downloads.arduino.cc/packages/package_index.json

clivepengelly, 2016-07-31:
Hi. I have installed a fresh copy of the IDE but continue to get this error when opening the boards manager. I have followed instructions on the forum to rectify the problem but nothing works.
I am running Windows 10.
I can access this URL from my browser.
I am able to install some of the boards that are listed in the boards manager, but most give errors when trying to install.
[image]

Error downloading http://downloads.arduino.cc/packages/package_index.json
java.lang.RuntimeException: java.lang.Exception: Error downloading http://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$0(ContributionManagerUI.java:151)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.Exception: Error downloading http://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:113)
	at cc.arduino.contributions.packages.ContributionInstaller.download(ContributionInstaller.java:330)
	at cc.arduino.contributions.packages.ContributionInstaller.downloadIndexAndSignature(ContributionInstaller.java:305)
	at cc.arduino.contributions.packages.ContributionInstaller.updateIndex(ContributionInstaller.java:283)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$0(ContributionManagerUI.java:147)
	... 1 more
Caused by: java.net.SocketException: Address family not supported by protocol family: connect
	at java.net.DualStackPlainSocketImpl.connect0(Native Method)
	at java.net.DualStackPlainSocketImpl.socketConnect(DualStackPlainSocketImpl.java:83)
	at java.net.AbstractPlainSocketImpl.doConnect(AbstractPlainSocketImpl.java:350)
	at java.net.AbstractPlainSocketImpl.connectToAddress(AbstractPlainSocketImpl.java:206)
	at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:188)
	at java.net.PlainSocketImpl.connect(PlainSocketImpl.java:172)
	at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
	at java.net.Socket.connect(Socket.java:589)
	at sun.net.NetworkClient.doConnect(NetworkClient.java:175)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:432)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:527)
	at sun.net.www.http.HttpClient.<init>(HttpClient.java:211)
	at sun.net.www.http.HttpClient.New(HttpClient.java:308)
	at sun.net.www.http.HttpClient.New(HttpClient.java:326)
	at sun.net.www.protocol.http.HttpURLConnection.getNewHttpClient(HttpURLConnection.java:1169)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect0(HttpURLConnection.java:1148)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect(HttpURLConnection.java:999)
	at sun.net.www.protocol.http.HttpURLConnection.connect(HttpURLConnection.java:933)
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:170)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:128)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:111)
	... 5 more

agdl, 2016-08-01:
Hi, If you open the preferences menu, you can navigate to the .Arduino15 folder clicking on the preferences path. Then open the folder staging/packages and try to delete all the archive you have there. Please let me know if the problem persist.

clivepengelly, 2016-08-01:
Hi.

Thanks for the reply.

Deleted the folder, same problem.

 

 

 

Regards,

Clive Pengelly

+27 82 882 4505

agdl, 2016-08-04:
Can you please give a try to this?

http://stackoverflow.com/questions/16373906/address-family-not-supported-by-protocol-family-socketexception-on-a-specific/21383865#21383865

clivepengelly, 2016-08-04:
GENIUS!

 

This worked.

 

I am surprised that many more people do not have this problem. I do not have anything special in my environment.

 

This is what I had to do – nothing else:-

 

1.    Open command prompt (windows start button, type cmd in search)

2.    Click top left of command prompt, click properties, enable quick edit mode

3.    Copy: setx _JAVA_OPTIONS -Djava.net.preferIPv4Stack=true

4.    Right click on command prompt, should paste

5.    Hit enter. That easy!

 

 

Regards,

Clive Pengelly

+27 82 882 4505

 

http://stackoverflow.com/questions/16373906/address-family-not-supported-by-protocol-family-socketexception-on-a-specific/21383865#21383865

cmaglie, 2016-08-04:
@clivepengelly 
from the linked article it seems that the problem is caused by a spyware:

>    Try to check whether the spy program called "RelevantKnowledge" is installed. Its
>     uninstallation helped me to solve the problem.

You may probably want to check if you have this program installed too.

BTW happy that you solved it!

closed 2016-08-04 by cmaglie

clivepengelly, 2016-08-04:
I do have this relevant knowledge thing running. What is it and where does it come from?

 

I download a lot of development and test software.

 

 

 

 

Regards,

Clive Pengelly

+27 82 882 4505

cmaglie, 2016-08-04:
It doesn't surely come from Arduino.
Probably you installed it bundled with some other software, without noticing.

From a quick search on google:
http://www.relevantknowledge.com/faq.aspx
http://www.shouldiremoveit.com/RelevantKnowledge-12964-program.aspx

sachin1389, 2016-08-19:
Please help the problem still exist for me 
i have done 
setx _JAVA_OPTIONS -Djava.net.preferIPv4Stack=true
Success: specified value was saved
and also checked for the relevant knowledge spy.

still getting error

 Error downloading http://downloads.arduino.cc/packages/package_index.json
java.lang.RuntimeException: java.lang.Exception: Error downloading http://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$19(ContributionManagerUI.java:151)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.Exception: Error downloading http://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:113)
	at cc.arduino.contributions.packages.ContributionInstaller.download(ContributionInstaller.java:330)
	at cc.arduino.contributions.packages.ContributionInstaller.downloadIndexAndSignature(ContributionInstaller.java:305)
	at cc.arduino.contributions.packages.ContributionInstaller.updateIndex(ContributionInstaller.java:283)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$19(ContributionManagerUI.java:147)
	... 1 more
Caused by: java.net.SocketTimeoutException: Connect timed out
	at java.net.SocksSocketImpl.readSocksReply(SocksSocketImpl.java:126)
	at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:459)
	at java.net.Socket.connect(Socket.java:589)
	at sun.net.NetworkClient.doConnect(NetworkClient.java:175)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:432)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:527)
	at sun.net.www.http.HttpClient.<init>(HttpClient.java:211)
	at sun.net.www.http.HttpClient.New(HttpClient.java:308)
	at sun.net.www.http.HttpClient.New(HttpClient.java:326)
	at sun.net.www.protocol.http.HttpURLConnection.getNewHttpClient(HttpURLConnection.java:1169)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect0(HttpURLConnection.java:1148)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect(HttpURLConnection.java:999)
	at sun.net.www.protocol.http.HttpURLConnection.connect(HttpURLConnection.java:933)
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:170)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:128)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:111)
	... 5 more
 Error downloading http://downloads.arduino.cc/packages/package_index.json
java.lang.RuntimeException: java.lang.Exception: Error downloading http://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$19(ContributionManagerUI.java:151)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.Exception: Error downloading http://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:113)
	at cc.arduino.contributions.packages.ContributionInstaller.download(ContributionInstaller.java:330)
	at cc.arduino.contributions.packages.ContributionInstaller.downloadIndexAndSignature(ContributionInstaller.java:305)
	at cc.arduino.contributions.packages.ContributionInstaller.updateIndex(ContributionInstaller.java:283)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$19(ContributionManagerUI.java:147)
	... 1 more
Caused by: java.net.SocketTimeoutException: Connect timed out
	at java.net.SocksSocketImpl.readSocksReply(SocksSocketImpl.java:126)
	at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:459)
	at java.net.Socket.connect(Socket.java:589)
	at sun.net.NetworkClient.doConnect(NetworkClient.java:175)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:432)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:527)
	at sun.net.www.http.HttpClient.<init>(HttpClient.java:211)
	at sun.net.www.http.HttpClient.New(HttpClient.java:308)
	at sun.net.www.http.HttpClient.New(HttpClient.java:326)
	at sun.net.www.protocol.http.HttpURLConnection.getNewHttpClient(HttpURLConnection.java:1169)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect0(HttpURLConnection.java:1148)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect(HttpURLConnection.java:999)
	at sun.net.www.protocol.http.HttpURLConnection.connect(HttpURLConnection.java:933)
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:170)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:128)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:111)
	... 5 more
 Error downloading http://downloads.arduino.cc/packages/package_index.json
java.lang.RuntimeException: java.lang.Exception: Error downloading http://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$19(ContributionManagerUI.java:151)
	at java.lang.Thread.run(Thread.java:745)
[... 29 lines cut ...]
Caused by: java.lang.Exception: Error downloading http://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:113)
[... cut ...]
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect0(HttpURLConnection.java:1148)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect(HttpURLConnection.java:999)
	at sun.net.www.protocol.http.HttpURLConnection.connect(HttpURLConnection.java:933)
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:170)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:128)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:111)
	... 5 more

sachin1389, 2016-08-19:
Hey Problem solved,I just tried remove as much software which was seeing to be unknown to me
and then deleted the arduino and reinstalled and then restarted my pc . now it is working fine.

adam4142, 2016-11-05:
Its worked ...........
you r a GENIUS>............

ArimoJak7, 2017-01-25:
thanks 
it worked

paulenuta, 2017-02-04:
I had same error, but could not manually open the http://downloads.arduino.cc/packages/package_index.json list and solve the problem be restarting PPOE connection to the ISP.

Technocratp, 2017-03-25:
it's a problem of Internet connectivity in PC or laptop.

bigelite, 2017-06-03:
i have this problem too, 
I have followed instructions on the forum to rectify the problem but nothing works.
I am running Windows 10.
I can access this URL from my browser.
I've opened the folder staging/packages and  deleted all the archive that i had.
I've tried this one: setx _JAVA_OPTIONS -Djava.net.preferIPv4Stack=true

but its still had an error. i don't know what to do!

ShotgunBurger, 2017-06-19:
@bigelite I have the same problem, Tried everything. Did you end up solving the issue?

ShotgunBurger, 2017-06-19:
I had this issue - I am running Windows 10, Arduino IDE 1.8.3.
I tried the CMD Prompt "setx _JAVA_OPTIONS -Djava.net.preferIPv4Stack=true
I tried a fresh install of the Arduino IDE
I tried resetting my PPPoE Connection via my modem to my ISP
I tried using WiFi, Ethernet and tethering to my cell phones ISP via personal hotspot
None of these fixed the problem.

I fixed my issue by going to "ThisPC>(C:)>Users>xxxx>AppData>Local>Arduino15" and deleting the file "package_index.json.tmp"

Now it works fine.

mrohner, 2017-06-28:
This setx _JAVA_OPTIONS -Djava.net.preferIPv4Stack=true
results in an error:
ERROR: Invalid syntax. Default option is not allowed more than '2' time(s).
Type "SETX /?" for usage.

jeanmoretto, 2017-07-01:
@ShotgunBurger perfect for me. I just deleted that file and "voilà".
It back to work.

hamid1234567, 2017-07-14:
I could solved this issue by correcting java configuration and change the proxy seeting of java network to correct solution. try these step in windows
1- search configure java  from start menu 
2- In general tab open network setting and correct it (i changed it to direct connection - no proxy )
arduino work and download rapidly any things

friesandacoke, 2017-07-30:
@ShotgunBurger That was what I needed. Perfect. Thanks!

To add, most will need to unhide hidden files and folders before they can see the appdata folders.

sunnydevtale, 2017-09-18:
Hi,

I am also facing the same error while managing the Arduino libraries. I have tried all above solution but this didn't worked for me.

For me, getting this error due to internet connectivity. I have switch to another internet network and tried again and this Works..!. 

Hopefully this issue will resolve for you also.

MrJ6, 2017-10-10:
ShotgunBurger, thanks man you are a life saver ^^

sunnydevtale, 2017-10-10:
:) YW..!

AlexSwensen, 2017-10-22:
I think it might be down today. Tried to access the URL from my phone and home connection. No dice.

AlexSwensen, 2017-10-22:
And its back.

tin345, 2017-11-18:
> I fixed my issue by going to "ThisPC>(C:)>Users>xxxx>AppData>Local>Arduino15" and deleting the file "package_index.json.tmp"

This one worked for me also.... Thanks Much...

VaidehiTrivedi, 2018-03-22:
Just try doing this 

I fixed my issue by going to "ThisPC>(C:)>Users>xxxx>AppData>Local>Arduino15" and deleting the file "package_index.json.tmp"

and if this doesn't work, refer this link : "https://stackoverflow.com/questions/16373906/address-family-not-supported-by-protocol-family-socketexception-on-a-specific/21383865#21383865"

haithamkob, 2018-07-11:
it worked for me by deleting  "ThisPC>(C:)>Users>xxxx>AppData>Local>Arduino15" and deleting the file "package_index.json.tmp"

Prateek7805, 2018-09-17:
it worked by deleted the Arduino15 folder contents now time to go back

pedrohas2000, 2018-09-20:
> @clivepengelly
> from the linked article it seems that the problem is caused by a spyware:
>
>
> You may probably want to check if you have this program installed too.
>
> BTW happy that you solved it!

It`s worked for me!

yokozawa766, 2018-09-27:
Just copy company's proxy setting to ArduinoIDE>preference>NetworkTab. It works. You can use library manager and board manager.

maurodom, 2018-10-08:
I tried every fix here with no results.
Afterwards I momentarely disabled my TinyWall firewall and then the "Error downloading http://downloads.arduino.cc/packages/package_index.json" disppeared.
All the boards I needed have been immediately downloaded.
Hope it helps.

yokozawa766, 2018-10-08:
MR. maurodom,
I write the way to the picture ;
http://yokozawa766.car.coocan.jp/proxy.jpg

zexalistic, 2018-10-23:
> I had this issue - I am running Windows 10, Arduino IDE 1.8.3
> I tried the CMD Prompt "setx _JAVA_OPTIONS -Djava.net.preferIPv4Stack=true"
> I tried a fresh install of the Arduino IDE
> I tried resetting [...]

Thank you very much!!!

cooldew2007, 2018-11-05:
> I had this issue - I am running Windows 10, Arduino IDE 1.8.3
> I tried the CMD Prompt "setx _JAVA_OPTIONS -Djava.net.preferIPv4Stack=true"
> I tried a fresh install of the Arduino IDE
> I tried resetting [...]

This worked for me. Cheers

DarkStar18, 2018-11-18:
@clivepengelly Wow, open the Command and paste "setx _JAVA_OPTIONS -Djava.net.preferIPv4Stack=true" WORK perfectly. Thank you so much for solving my problem

xuanbka1, 2019-03-06:
> I had this issue - I am running Windows 10, Arduino IDE 1.8.3
> I tried the CMD Prompt "setx _JAVA_OPTIONS -Djava.net.preferIPv4Stack=true"
> I tried a fresh install of the Arduino IDE
> I tried resetting [...]

thanks !! so lady

AnnaLobo, 2019-03-26:
> I had this issue - I am running Windows 10, Arduino IDE 1.8.3
> I tried the CMD Prompt "setx _JAVA_OPTIONS -Djava.net.preferIPv4Stack=true"
> I tried a fresh install of the Arduino IDE
> I tried resetting [...]

thaaaaanks the only thing that worked \o/

amrithmmh, 2019-04-30:
> I had this issue - I am running Windows 10, Arduino IDE 1.8.3
> I tried the CMD Prompt "setx _JAVA_OPTIONS -Djava.net.preferIPv4Stack=true"
> I tried a fresh install of the Arduino IDE
> I tried resetting [...]

this worked for me

MattCosturos, 2019-05-03:
This issue is still happening!

I had to go to https://downloads.arduino.cc/packages/package_index.json and then download all the certificates.
Then I had to manually import these certs into the cacerts file in the JRE which was installed in the Arduino directory! (not in the global java directory!)

Arduino installs it's own jre in C:\Program Files (x86)\Arduino\java\ and that cacerts file needs to be updated!

Kind of a shame that the currently shipping version has a jre with missing certs needed to download the board packages.

NeMeSiS97, 2019-05-10:
> I had this issue - I am running Windows 10, Arduino IDE 1.8.3
> I tried the CMD Prompt "setx _JAVA_OPTIONS -Djava.net.preferIPv4Stack=true"
> I tried a fresh install of the Arduino IDE
> I tried resetting [...]

This is indeed the correct solution.
Thanks alot mate

iofirag, 2019-06-24:
open arduino ide
File > Preferences

copy paste to the input box of (Additional boards manager urls):
https://arduino.esp8266.com/stable/package_esp8266com_index.json

click on arduino folder url link at the bottom. and leave it aside.

click ok in preferences screen.

open cmd and copy paste this:
"setx _JAVA_OPTIONS -Djava.net.preferIPv4Stack=true"

open ide again, and wait a minute.
looking on the content folder from before, when something has changed,
wait some seconds to finish, 
and then, 
delete all from the folder except from the preferences.txt

open:
Tools > Board > Board Manager

see if the loader in the bottom finish working

smtpsantiago, 2019-07-24:
They recommend deleting the file package_index.json that is stored in package_index.json or package_index.json.tmp

JeffrySan, 2019-09-05:
I had this issue - I am running Windows 10, Arduino IDE 1.8.3
I tried the CMD Prompt "setx _JAVA_OPTIONS -Djava.net.preferIPv4Stack=true"
I tried a fresh install of the Arduino IDE
I tried resetting my PPPoE Connection via my modem to my ISP
I tried using WiFi, Ethernet and tethering to my cell phone's ISP via personal hotspot
I tried by going to "ThisPC>(C:)>Users>xxxx>AppData>Local>Arduino15" and delete all of them and nothing happend
None of these fixed the problem.

but when i try to uninstall java and reinstall it and plus to configure java path on Advance System Settings, it's good (in condition i reinstall arduino, i delete file in arduino>file>preferences folder)

Goodluck guyss

pvinod424, 2019-09-13:
It worked for me!! Try this definately
I fixed my issue by going to "ThisPC>(C:)>Users>xxxx>AppData>Local>Arduino15" and deleting the file "package_index.json.tmp"

Ksol13, 2019-10-09:
I have the same problem.
But if i try to delete "package_index.json.tmp", a new file comes up again and again.
Whats my mistake?

JeffrySan, 2019-10-11:
Re Install your Java from computer :
1. Uninstall java from (add remove program (search))
2. Install Java from https://www.java.com/en/download/

MiracelVip, 2020-01-03:
I have the same problem here... my computer is connected to an IPv6 DSLITE internet. I cant open the "https://downloads.arduino.cc/libraries/library_index.json.sig" it says "not found" the only posibility was to open a hotspot with my mobile and download the file

It seems like a ipv4 ipv6 problem

yihanglim, 2020-06-09:
> I have the same problem.
> But if i try to delete "package_index.json.tmp", a new file comes up again and again.
> Whats my mistake?

Same here, tried all method stated above, the file just keep comes up again when try to open board manager. Are you able to solve it yet?

ptrooms, 2020-06-09:
I've perhaps a related problem.
When I try to use Board_manager, I run into an java.lang.RuntimeException: 
```
java.lang.NullPointerException
java.lang.RuntimeException: java.lang.NullPointerException
        at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$1(ContributionManagerUI.java:150)
        at java.lang.Thread.run(Thread.java:748)
```
After this, Arduino cannot be restarted as long the "package_index.json" exists. 
Removing this re-iterates the problem as soon as I (need to) update board manager.

Note: I run Linux, the problem initiated on Arduino 18.5 and persists with the latest 1.8.12. The same happens on a clean  machine. Hence, something is wrong with "package_index.json"

KTibow, 2020-06-09:
You could try deleting all files that seem related to `package_index.json`, not just `package_index.json.tmp`.

ptrooms, 2020-06-09:
@KTibow : same result. Clearing the .Arduino15 directory will refill it with newly downloaded files as soon when I try to use BoardManager.

yihanglim, 2020-06-10:
@ptrooms after spending whole night of troubleshooting, I couldn't find a solution. This version used to work fine for a few days before and suddenly this issue occurred out of nothing. So I uninstalled 1.8.12 and installed the hourly builds (last upated 3 June 2020 22:13:17 GMT) . So far everything runs alright.

ptrooms, 2020-06-10:
@yihanglim , yes metoo. Currently I'm using the nightly version also. See the other threads for more details. Time for a good nap of sleep.

facchinm, 2020-06-10:
We had a problem on our package_index.json that caused the error and prevented the IDE to start.
Now the package_index has been fixed, but to recover a working installation you must delete once again the package_index.json with the cache folder and restart the IDE as usual.

Windows:
* remove file C:\Users\YourUsername\AppData\Local\Arduino15\package_index.json
* remove folder C:\Users\YourUsername\AppData\Local\Arduino15\cache

Mac:
* remove file /Users/YourUsername/Library/Arduino15/package_index.json
* remove folder /Users/YourUsername/Library/Arduino15/cache

Linux:
* remove file /home/YourUsername/.arduino15/package_index.json
* remove folder /home/YourUsername/.arduino15/cache

Please note that the problem that prevents the IDE to start has already been fixed and it's ready for the next release, that's the reason why the Nightly/Beta Builds are not affected, and also the reason why we did not detect this problem earlier, sorry about that!

dharmik768, 2021-05-05:
Tried everything and at last, @ShotgunBurger fixed my bug thanks man.

dammyjay, 2021-09-16:
> > I fixed my issue by going to "ThisPC>(C:)>Users>xxxx>AppData>Local>Arduino15" and deleting the file "package_index.json.tmp"
> 
> This one worked for me also.... Thanks Much...

so reliefed

bunya017, 2022-07-15:
If everything here doesn't work for you, download the package manually and place it in `C:\Users\< USER >\AppData\Local\Arduino15\staging\packages`. Then go back to the IDE and install again.

behnamsaberi, 2023-01-26:
> I had this issue - I am running Windows 10, Arduino IDE 1.8.3 I tried the CMD Prompt "setx _JAVA_OPTIONS -Djava.net.preferIPv4Stack=true" I tried a fresh install of the Arduino IDE I tried resetting [...]

hi , it worked for me too
thanksssss