arduino/Arduino #5189: Upload issue

johbell, 2016-07-31:
After updating to 1.6.10 I can not communicate with my MKS Base 1.3 board. When I try to upload the firmware it just keeps saying uploading with no activity on the board. I went back to the 1.6.9 version and everything worked fine.

meoudon, 2016-08-01:
I have solved. It Avrdude's error. I copy worked avrdude from another version into /avr/bin and avrdude.conf. Everything done on 1.6.12.

harryboo, 2016-08-03:
Please test this [build](https://github.com/arduino/Arduino/pull/5202) !

per1234, 2016-08-03:
> Please test this build !

Unless there were changes in avrdude other than the avrdude.conf lock bits read mask that build should only affect issues related to **Tools > Burn Bootloader**.

cmaglie, 2016-09-22:
Should be fixed with #5374 and released in IDE 1.6.12.
Please reopen if the issue is not solved.

closed 2016-09-22 by cmaglie