arduino/Arduino #5191: "Save as..." Dialog box freezes entire IDE

jabelone, 2016-08-01:
I noticed this bug a long time ago but it's still present.  This is almost a fresh install of windows 10 and I'm fairly sure it used to do it with Windows 8.  Basically, whenever you have the save as window open and try to delete a file/folder, the entire IDE hangs.

Although it hangs, it does actually delete the file/folder though.  When I say hang, the entire IDE becomes un-responsive and must be forcefully closed.  It's a small bug but annoying if I forget and it happens.

Other info:
Operating System: Windows 10 (64 bit)
IDE Version: 1.6.10
Completely repeatable, happens 100% of the time

thijmenketel, 2017-03-05:
I can confirm that this bug still exists. Running the 1.8.2 IDE.

thijmenketel, 2017-03-06:
When I did some googling, i found that the bug probably resides in the `java.awt.FileDialog`. More users found this [here](https://bugs.openjdk.java.net/browse/JDK-8161467).

According to [this commit](http://hg.openjdk.java.net/jdk9/jdk9/jdk/rev/fcbe313c8cec), the bug is fixed in JDK-9. (if im right, dont know that for sure)

slootsky, 2018-03-04:
I'm experiencing the same issue with Arduino 1.6.8 or 1.8.4 on Windows 10 running Java 8 Update 161

bobwmcgrath, 2018-08-18:
I am also having this problem