arduino/Arduino #5195: Linux Mint vs. Arduino IDE/ArduBlock translations

rpgmem, 2016-08-01:
Gentlemen

I'm running the IDE Arduino on Linux Mint (Debian Edition), however I found a small bug that needs to be investigated.

When installing the program (using Synaptic) and open it the system language is not recognized automatically (my system default language is pt_BR) and the program loads in English.

Until then no major problems, did I install the newest version of ArduBlock to compose the tools that I use with some students.

That's where the problems started. To manually translate the Arduino IDE Ardublock is completely in English. Returning the IDE Arduino to "default system language" Arduino IDE is in English already ArduBlock begins to appear in Portuguese.

It seems that the IDE Arduino is making a huge mess in the recognition system's default language and passing the wrong information to ArduBlock.

rpgmem, 2016-08-14:
it seems that the problem is in how the Arduino IDE is to manage the language. It does not use the default naming example: en_EN / pt_BR etc.
It uses the naming en_en / pt_br which in the case of Linux are two completely different things.

Could check and effect the correction? Grateful!

cmaglie, 2016-08-15:
@rpgmem 

The language list is defined here:
https://github.com/arduino/Arduino/blob/master/app/src/cc/arduino/i18n/Languages.java

for English we have two variants:
```java
      new Language(tr("English"), "English", "en"),
      new Language(tr("English (United Kingdom)"), "English (United Kingdom)", "en_GB"),
```

for Portugese:
```java
      new Language(tr("Portugese"), "Português", "pt"),
      new Language(tr("Portuguese (Brazil)"), "Português (Brazil)", "pt_BR"),
      new Language(tr("Portuguese (Portugal)"), "Português (Portugal)", "pt_PT"),
```

so it seems that the IDE use the standarnd naming, how do you obtain the "en_en"?

For completeness here the resources:
https://github.com/arduino/Arduino/tree/master/arduino-core/src/processing/app/i18n

And here the i18n helper functions:
https://github.com/arduino/Arduino/tree/master/arduino-core/src/processing/app/i18n

per1234, 2021-03-27:
Closing due to lack of response.

@rpgmem if you'll provide the requested information I'll be happy to re-open this issue.

closed 2021-03-27 by per1234