arduino/Arduino #5200: www.arduino.cc/en/Reference/Int add: Arduino Zero

avandalen, 2016-08-03:
Change:
On the Arduino Due, an int stores a 32-bit (4-byte) value
to
On the Arduino Due and Zero, an int stores a 32-bit (4-byte) value

q2dg, 2016-08-03:
Well...I remember this https://github.com/arduino/Arduino/issues/4525 (and this https://github.com/arduino/Arduino/issues/3801)

cousteaulecommandant, 2016-09-02:
...And while we're at it, fix the signed overflow thing (#4511) (either remove that part from the documentation or make Arduino actually do what the documentation says)

cmaglie, 2017-01-26:
Actually the page says:

> On the Arduino Due and SAMD based boards (like MKR1000 and Zero),

I don't know if there the page has been edited in the meantime.

closed 2017-01-26 by cmaglie

avandalen, 2017-01-26:
Dear Cristian

I will help you but I don’t know which article you refer?

Albert

From: Cristian Maglie 
Sent: Thursday, January 26, 2017 9:39 PM
To: arduino/Arduino 
Cc: Albert van Dalen ; Author 
Subject: Re: [arduino/Arduino] www.arduino.cc/en/Reference/Int add: Arduino Zero (#5200)

Actually the page says:

  On the Arduino Due and SAMD based boards (like MKR1000 and Zero),

I don't know if there the page has been edited in the meantime.