arduino/Arduino #5203: Multiplication Error

birotx, 2016-08-03:
We have the following piece of code:

`    byte msb = data[1];
    byte lsb = data[0];
    int result = (msb << 8) | lsb;
    long RT = result*100;
    RT=RT;
    Serial.print(result);
    Serial.print(" ");
    Serial.print(RT);
    Serial.println(" ");`

and this is a line of the serial: 
`404 -25136`

So 404*100 is  -25136

Then we just move -*100 from the declaration of RT variable to the next line like following:
`    byte msb = data[1];
    byte lsb = data[0];
    int result = (msb << 8) | lsb;
    long RT = result;
    RT=RT*100;
    Serial.print(result);
    Serial.print(" ");
    Serial.print(RT);
    Serial.println(" ");`

and a line from serial:

`404 40400 `

Everything works like a charm! I spent hours to figure what was wrong....

per1234, 2016-08-03:
Since your issue is resolved please close it.

birotx, 2016-08-04:
@per1234 My issue is solved because I moved the multiplication one row below, a quick fix, but it is not ok that arduino can't multiply on the declaration of a variable, at least this is what is happening to me...

peabotai, 2016-08-04:
The problem has nothing to do with Arduino.  It is the way the C++ language works.  When you write

 long RT = result*100;

the rules for arithmetic in C++ specify that the multiplication result*100 is carried out as int, but the assignment then converts the int to a long.

Here is an easy way to avoid this:

long RT = result*100L;

or if you were using two variables instead of a a variable and a constant:

long RT = result*(long)scale;

The C++ rules then say that result must be converted to a long before being multiplied because multiplying an int by a long requires both terms to be long.

The C++ rules for arithmetic are sometimes surprising.  Another thing to be cautious about is combining signed and unsigned quantities.

Peter Olson

matthijskooijman, 2016-08-04:
As @peabotai has shown, this is indeed how the language works, so I'm closing this issue. If you think better documentation could have prevented your confusion, feel free to add a comment suggestion where to improve the documentation.

closed 2016-08-04 by matthijskooijman