arduino/Arduino #5205: Arduino ISP on IDE 1.6.10 do not work

logicaprogrammabile, 2016-08-04:
Hi
i have tested the ArduinoISP with Genuino UNO R3 with the new IDE 1.6.10 but is not possible burning the bootloader in new atmega328p. 
this is the error message:

`
avrdude: Version 6.3, compiled on Jun 22 2016 at 16:05:21
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM9
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 Version 1.x firmware
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.03s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: erasing chip
avrdude: reading input file "0x3F"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x3F:
avrdude: load data lock data from input file 0x3F:
avrdude: input file 0x3F contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.02s

avrdude: verifying ...
avrdude: 1 bytes of lock verified
avrdude: reading input file "0x05"
avrdude: writing efuse (1 bytes):

Writing | Errore durante la scrittura del bootloader
 ***failed;  
################################################## | 100% 0.11s

avrdude: 1 bytes of efuse written
avrdude: verifying efuse memory against 0x05:
avrdude: load data efuse data from input file 0x05:
avrdude: input file 0x05 contains 1 bytes
avrdude: reading on-chip efuse data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
         0xfd != 0x05
avrdude: verification error; content mismatch

avrdude done.  Thank you.

`

awatterott, 2016-08-04:
There is already issue reports about the problem here:
https://github.com/arduino/Arduino/issues/5175
https://github.com/arduino/avrdude-build-script/issues/2
You have to modify avrdude.conf and boards.txt to get it working.

per1234, 2016-08-04:
This issue is only #5175. I don't think arduino/avrdude-build-script#2, which would have caused the lock to not verify, affects ArduinoISP. You should be able to fix it by making the changes shown in https://github.com/arduino/Arduino/pull/5182 or you can try the test build https://github.com/arduino/Arduino/pull/5202#issuecomment-237621343 which should also fix it in a different way but I haven't had a chance to try it yet. If you do test it please let us know your results.

logicaprogrammabile, 2016-08-05:
after ide 1.6.10 installation, i have replace the boards.txt with the files shown in #5182. 
I have tested the burning bootloader with atmega328p and attiny85 and in both case the ide successfully burn it.
Thank you.

cmaglie, 2016-09-29:
@logicaprogrammabile is this problem is fixed in IDE 1.6.12? Can we close this issue?

cmaglie, 2017-01-19:
Closing as solved, reopen if the issue persists.

closed 2017-01-19 by cmaglie