arduino/Arduino #5206: Error uploading sketch in 1.6.10:  java.io.IOException: Cannot run program "___REMOVE___/bin/avrdude": error=2, No such file or directory

q2dg, 2016-08-04:
When I want to upload any sketch to my Arduino UNO, I get this error:

_java.io.IOException: Cannot run program "___REMOVE___/bin/avrdude": error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at java.lang.Runtime.exec(Runtime.java:620)
	at java.lang.Runtime.exec(Runtime.java:485)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:11)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:130)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:209)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.Sketch.upload(Sketch.java:1187)
	at processing.app.Sketch.exportApplet(Sketch.java:1160)
	at processing.app.Sketch.exportApplet(Sketch.java:1132)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2409)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:248)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 11 more
An error occurred while uploading the sketch_

I'm using Fedora 24 64bits

q2dg, 2016-08-04:
With 1.6.9 works fine

q2dg, 2016-08-04:
Oh, sorry, it was a problem of permissions. Sorry.

closed 2016-08-04 by q2dg

Apollon77, 2016-08-04:
What permission was incorrect? I have the same problem on macOS at 1.6.10

q2dg, 2016-08-04:
@Apollon77 Well, I just did _chmod 777 /run/lock_  but it's a little strange because I already did this before, when downloading former versions of the IDE

davidbitton, 2016-08-13:
I'm SOL too. 1.6.10 macOS 10.11.6

```
bittondb:~/ $ sudo chmod 777 /run/lock                                                                                               [0:21:47]
chmod: /run/lock: No such file or directory
```

alexander-yun, 2016-08-13:
Same problem as topic starter with arduino v. 1.6.10 and macOS 10.11.6

alexander-yun, 2016-08-13:
On running an arduion v 1.6.10 it auto update an avr library up to version 1.6.12 and all work