arduino/Arduino #5209: <cstddef> missing from 1.6.9 thru 1.6.12

moxondesign, 2016-08-05:
Error message 

C:\Users\moxon\Documents\Arduino\libraries\Embedis\src/Embedis.h:33:19: fatal error: cstddef: No such file or directory

 #include <cstddef>

                   ^

compilation terminated.

Using library Embedis at version 1.1.0 in folder: C:\Users\moxon\Documents\Arduino\libraries\Embedis 
exit status 1
Error compiling for board Arduino Pro or Pro Mini.

To duplicate : Install Library "Embedis" using the Library Manager and compile the example program.

Tested under Windows7 & Debian 3.16.7 (x64)

matthijskooijman, 2016-08-05:
I think cstdef, just like pretty much all of libstdc++ has been missing for AVR-based Arduino forever. Or are you suggesting that this used to work before?

One way to get that file is, I think, the StandardCplusplus library. Another way is to just include "stddef.h", which I think has the same contents.

moxondesign, 2016-08-06:
> Or are you suggesting that this used to work before?
Yes, this worked fine up to Arduino AVR Board version 1.6.8 - then broken in 1.6.9 and thereafter...
(just a few months ago....)
This works just fine on most other platforms ESP8266, Due, SAMD, etc. -
not sure why this should regress with a later version of the AVR compiler?
library :Embedis is a good regression test case for this perhaps

oqibidipo, 2016-08-07:
> this worked fine up to Arduino AVR Board version 1.6.8 - then broken in 1.6.9

This is when `-std=gnu11` and `-std=gnu++11` were added to compiler flags.
`Embedis.h` does different things depending on the C++ version:
```
// Arduino is like a box of chocolates.
// You never know what you're going to get.
#if __cplusplus < 201103L
#include <stddef.h>
#include <stdint.h>
inline void * operator new (size_t size, void * ptr) {
    (void)size;
    return ptr;
}
#else
#include <cstddef>
#include <cstdint>
#include <new>
#endif
```
Conclusion: `<cstddef>` has not suddenly disappeared in AVR boards 1.6.9, it was never there in the first place. Embedis should make more detailed version and platform checks.

matthijskooijman, 2016-08-07:
@oqibidipo Ah, that explains things indeed. Thanks for digging into this. I think this is not a problem in Arduino then, so I'm closing this issue. I'll also invite @moxondesign to report this bug to the Embedis maintainers.

closed 2016-08-07 by matthijskooijman