arduino/Arduino #5210: Crash on IDE startup

storzinc, 2016-08-05:
Arduino 1.6.10 is crashing on startup on my Windows 7, Arduino 1.6.7 is working fine on the same machine. It is crashing here:
https://github.com/facchinm/libserialport/blob/70af63f12ae290a20a05e71d933d72d0db85b93a/windows.c#L272

port->usb_path contains "2" there.

facchinm, 2016-08-05:
Hi, could you post the specs of your PC and the attached boards/serial interfaces? Thanks

storzinc, 2016-08-06:
My PC: CPU - Intel Q9550, Motherboard - Asus P5Q-E, RAM - 8GB

The board is Iskra Neo, it is third party clone of the Arduino Leonardo, in the Device Manager it is displayed as "Arduino Leonardo (COM7)",  Hardware ID: USB\VID_2341&PID_8036&REV_0100

facchinm, 2016-08-10:
Thanks for the info, the board itself shouldn't be the problem.
Could you try running `listSerialC.exe` from [this](http://downloads.arduino.cc/liblistSerials/liblistSerials-1.1.4.zip) package from a `cmd` shell after setting LIBSERIALPORT_DEBUG environment variable with
```
set LIBSERIALPORT_DEBUG=1
```
Please post here the result so I'll take care of debugging the problem (which could affect other people as well)

storzinc, 2016-08-10:
`listSerialC.exe` is also crashing on my machine, before the crash it prints this:

```
sp: sp_list_ports(0028FEA4) called.
sp: Enumerating ports.
sp: Opening registry key.
sp: Querying registry key value and data sizes.
sp: Iterating over values.
sp: Found port COM1.
sp: sp_get_port_by_name_desc(COM1, 003C4FD0) called.
sp: Building structure for port COM1.
sp: get_port_details returning SP_OK.
sp: sp_get_port_by_name_desc returning SP_OK.
sp: Found port COM7.
sp: sp_get_port_by_name_desc(COM7, 003C4FD4) called.
sp: Building structure for port COM7.
```

danhusan, 2016-09-17:
Same issue;

Windows 7, device manager is showing the following ports:
Intel AMT (Com3)
Sierra Wireless MC8355 Gobi(TM) 3000 USB Diagnostics (Com41)
Sierra Wireless MC8355 Gobi(TM) 3000 NMEA (Com18)
Standard wireless over Bluetooth link (Com37)

C:\Program Files (x86)\Arduino\lib>listSerialC.exe
sp: sp_list_ports(0028FEA4) called.
sp: Enumerating ports.
sp: Opening registry key.
sp: Querying registry key value and data sizes.
sp: Iterating over values.
sp: Found port COM3.
sp: sp_get_port_by_name_desc(COM3, 00601970) called.
sp: Building structure for port COM3.
sp: get_port_details returning SP_OK.
sp: sp_get_port_by_name_desc returning SP_OK.
sp: Found port COM37.
sp: sp_get_port_by_name_desc(COM37, 00601BCC) called.
sp: Building structure for port COM37.
sp: get_port_details returning SP_OK.
sp: sp_get_port_by_name_desc returning SP_OK.
sp: Found port COM41.
sp: sp_get_port_by_name_desc(COM41, 00601BD0) called.
sp: Building structure for port COM41.

danhusan, 2016-10-03:
I can verify that removing the 'Sierra Wireless' card from my machine stops the IDE from crashing. Any progress troubleshooting?

fototakas, 2016-11-27:
I have similar issue, but on MAC OSX 10.11.6. Arduino IDE crash with installed rebranded 'Sierra Wireless' GSM modem. If I remove it, IDE starts without problem. But with this card, starting from version 1.6.8 Arduino IDE crash on startup with this exception:
```
Process:               Arduino [632]
Path:                  /Users/USER/Downloads/Arduino.app/Contents/MacOS/Arduino
Identifier:            cc.arduino.Arduino
Version:               1.6.13 (1.6.13)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Arduino [632]
User ID:               501

Date/Time:             2016-11-27 20:34:46.902 +0200
OS Version:            Mac OS X 10.11.6 (15G31)
Report Version:        11
Anonymous UUID:        CA5CA1E2-4261-0A0C-CE61-220A5B55CE2A

Time Awake Since Boot: 2300 seconds

System Integrity Protection: disabled

Crashed Thread:        27  Java: cc.arduino.packages.discoverers.serial.SerialBoardsLister

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[__NSCFString _getValue:forType:]: unrecognized selector sent to instance 0x7f95cd93a6e0'
terminating with uncaught exception of type NSException
abort() called

Application Specific Backtrace 1:
0   CoreFoundation                      0x00007fff93cf74f2 __exceptionPreprocess + 178
1   libobjc.A.dylib                     0x00007fff871a873c objc_exception_throw + 48
2   CoreFoundation                      0x00007fff93d611ad -[NSObject(NSObject) doesNotRecognizeSelector:] + 205
3   CoreFoundation                      0x00007fff93c67571 ___forwarding___ + 1009
4   CoreFoundation                      0x00007fff93c670f8 _CF_forwarding_prep_0 + 120
5   CoreFoundation                      0x00007fff93c08ecf CFNumberGetValue + 287
6   liblistSerialsj.dylib               0x0000000123b1ca13 get_port_details + 2243
7   liblistSerialsj.dylib               0x0000000123b1d626 sp_get_port_by_name_desc + 678
8   liblistSerialsj.dylib               0x0000000123b1e5b1 list_append + 145
9   liblistSerialsj.dylib               0x0000000123b1d0b8 list_ports + 728
10  liblistSerialsj.dylib               0x0000000123b1e88c sp_list_ports + 300
11  liblistSerialsj.dylib               0x0000000123b1bda0 Java_processing_app_Platform_listSerialsNative + 96
12  ???                                 0x000000010eb009f4 0x0 + 4541385204
13  ???                                 0x000000010eaf1040 0x0 + 4541321280
```

facchinm, 2016-11-28:
@danhusan is the issue still there on IDE 1.6.13?
@fototakas it looks like one of the call to `CFNumberGetValue` fails badly, probably one of these https://github.com/arduino/libserialport/blob/master/macosx.c#L144 . Can you get the VID/PID of the Sierra board from the Device Info on your Mac? Also,if you could launch the IDE from commandline as  `LIBSERIALPORT_DEBUG=1  /Applications/Arduino.app/Contents/Java/arduino`  and paste here the result it would be great

fototakas, 2016-11-28:
@facchinm Thanks for fast response.
Sierra board is rebranded as DELL DW5550:
```
DW5550:
  Product ID:	0x818d
  Vendor ID:	0x413c  (Dell Inc.)
  Version:	0.00
  Serial Number:	482AA32274503000
  Speed:	Up to 480 Mb/sec
  Manufacturer:	Dell
  Location ID:	0x1d160000 / 4
  Current Available (mA):	1000
  Current Required (mA):	0
  Extra Operating Current (mA):	0
  Built-In:	Yes
```

Debug enabled Arduino start log:
```
sp: sp_list_ports(0x7000016894d0) called.
sp: Enumerating ports.
sp: Creating matching dictionary.
sp: Getting matching services.
sp: Iterating over results.
sp: Found port /dev/cu.pci-serial22.
sp: sp_get_port_by_name_desc(/dev/cu.pci-serial22, 0x7ff28cff20e0) called.
sp: Building structure for port /dev/cu.pci-serial22.
sp: Getting serial port list.
sp: Iterating over results.
sp: Found port /dev/cu.pci-serial22.
sp: Found description PCI Serial Adapter (22).
sp: get_port_details returning SP_OK.
sp: sp_get_port_by_name_desc returning SP_OK.
sp: Found port /dev/cu.Bluetooth-Incoming-Port.
sp: sp_get_port_by_name_desc(/dev/cu.Bluetooth-Incoming-Port, 0x7ff28e106958) called.
sp: Building structure for port /dev/cu.Bluetooth-Incoming-Port.
sp: Getting serial port list.
sp: Iterating over results.
sp: Found port /dev/cu.Bluetooth-Incoming-Port.
sp: Found description Bluetooth-Incoming-Port.
sp: get_port_details returning SP_OK.
sp: sp_get_port_by_name_desc returning SP_OK.
sp: Found port /dev/cu.wwan.
sp: sp_get_port_by_name_desc(/dev/cu.wwan, 0x7ff28e106960) called.
sp: Building structure for port /dev/cu.wwan.
sp: Getting serial port list.
sp: Iterating over results.
sp: Found port /dev/cu.wwan.
sp: Found description DW5550.
2016-11-28 17:50:39.531 Arduino[1146:82472] -[__NSCFString _getValue:forType:]: unrecognized selector sent to instance 0x7ff28e1069c0
2016-11-28 17:50:39.532 Arduino[1146:82472] Apple AWT Internal Exception: -[__NSCFString _getValue:forType:]: unrecognized selector sent to instance 0x7ff28e1069c0
2016-11-28 17:50:39.532 Arduino[1146:82472] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[__NSCFString _getValue:forType:]: unrecognized selector sent to instance 0x7ff28e1069c0'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff93cf74f2 __exceptionPreprocess + 178
	1   libobjc.A.dylib                     0x00007fff871a873c objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff93d611ad -[NSObject(NSObject) doesNotRecognizeSelector:] + 205
	3   CoreFoundation                      0x00007fff93c67571 ___forwarding___ + 1009
	4   CoreFoundation                      0x00007fff93c670f8 _CF_forwarding_prep_0 + 120
	5   CoreFoundation                      0x00007fff93c08ecf CFNumberGetValue + 287
	6   liblistSerialsj.dylib               0x00000001223a7a13 get_port_details + 2243
	7   liblistSerialsj.dylib               0x00000001223a8626 sp_get_port_by_name_desc + 678
	8   liblistSerialsj.dylib               0x00000001223a95b1 list_append + 145
	9   liblistSerialsj.dylib               0x00000001223a80b8 list_ports + 728
	10  liblistSerialsj.dylib               0x00000001223a988c sp_list_ports + 300
	11  liblistSerialsj.dylib               0x00000001223a6da0 Java_processing_app_Platform_listSerialsNative + 96
	12  ???                                 0x000000010d3699f4 0x0 + 4516649460
	13  ???                                 0x000000010d35a040 0x0 + 4516585536
)
libc++abi.dylib: terminating with uncaught exception of type NSException
Abort trap: 6
```

fototakas, 2016-12-05:
@facchinm 
Hello Martino, any update on this issue?

facchinm, 2016-12-05:
Hi @fototakas ,
no update yet, we are still investigating with particular hardware but are unable to reproduce the issue. As per Apple documentation, `IORegistryEntrySearchCFProperty` should return `NULL` in case of error, so https://github.com/arduino/libserialport/blob/master/macosx.c#L121 should return before executing the wrong instruction. Can I send you some heavy debug binaries to understand what is going on?

facchinm, 2016-12-05:
I'm attaching the binary, if you could run it as 
```bash
LIBSERIALPORT_DEBUG=1 osx/listSerialC
```
and paste here the result it would be great (yes, it should crash nevertheless in your configuration)
[osx.zip](https://github.com/arduino/Arduino/files/631739/osx.zip)

fototakas, 2016-12-05:
Here is the results LIBSERIALPORT_DEBUG=1 ./listSerialC 

```
sp: sp_list_ports(0x7fff5fbffaa0) called.
sp: Enumerating ports.
sp: Creating matching dictionary.
sp: Getting matching services.
sp: Iterating over results.
sp: Found port /dev/cu.Bluetooth-Incoming-Port.
sp: sp_get_port_by_name_desc(/dev/cu.Bluetooth-Incoming-Port, 0x1001034d0) called.
sp: Building structure for port /dev/cu.Bluetooth-Incoming-Port.
sp: Getting serial port list.
sp: Iterating over results.
sp: Found port /dev/cu.Bluetooth-Incoming-Port.
sp: Found description Bluetooth-Incoming-Port.
Segmentation fault: 11
```

facchinm, 2016-12-05:
Ok, I'm producing a new one with less debug (this one contained so much it crashed for other reasons :smile: )

facchinm, 2016-12-05:
Updated version with one extra check (it shouldn't crash this time)
If you can, also try replacing the .dylib into the IDE installation and check if it solves your issue
[osx.zip](https://github.com/arduino/Arduino/files/631937/osx.zip)

fototakas, 2016-12-05:
Sorry, but no luck and this time :(

LIBSERIALPORT_DEBUG=1 ./listSerialC

```
sp: sp_list_ports(0x7fff5fbffaa0) called.
sp: Enumerating ports.
sp: Creating matching dictionary.
sp: Getting matching services.
sp: Iterating over results.
sp: Found port /dev/cu.Bluetooth-Incoming-Port.
sp: sp_get_port_by_name_desc(/dev/cu.Bluetooth-Incoming-Port, 0x1001034d0) called.
sp: Building structure for port /dev/cu.Bluetooth-Incoming-Port.
sp: Getting serial port list.
sp: Iterating over results.
sp: Found port /dev/cu.Bluetooth-Incoming-Port.
sp: Found description Bluetooth-Incoming-Port.
sp: get_port_details returning SP_OK.
sp: sp_get_port_by_name_desc returning SP_OK.
sp: Found port /dev/cu.pci-serial22.
sp: sp_get_port_by_name_desc(/dev/cu.pci-serial22, 0x100400038) called.
sp: Building structure for port /dev/cu.pci-serial22.
sp: Getting serial port list.
sp: Iterating over results.
sp: Found port /dev/cu.pci-serial22.
sp: Found description PCI Serial Adapter (22).
sp: get_port_details returning SP_OK.
sp: sp_get_port_by_name_desc returning SP_OK.
sp: Found port /dev/cu.wwan.
sp: sp_get_port_by_name_desc(/dev/cu.wwan, 0x100400040) called.
sp: Building structure for port /dev/cu.wwan.
sp: Getting serial port list.
sp: Iterating over results.
sp: Found port /dev/cu.wwan.
sp: Found description DW5550.
2016-12-05 20:19:51.256 listSerialC[7101:1402164] -[__NSCFString _getValue:forType:]: unrecognized selector sent to instance 0x100600000
2016-12-05 20:19:51.261 listSerialC[7101:1402164] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[__NSCFString _getValue:forType:]: unrecognized selector sent to instance 0x100600000'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff9a1234f2 __exceptionPreprocess + 178
	1   libobjc.A.dylib                     0x00007fff8d5d473c objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff9a18d1ad -[NSObject(NSObject) doesNotRecognizeSelector:] + 205
	3   CoreFoundation                      0x00007fff9a093571 ___forwarding___ + 1009
	4   CoreFoundation                      0x00007fff9a0930f8 _CF_forwarding_prep_0 + 120
	5   CoreFoundation                      0x00007fff9a034ecf CFNumberGetValue + 287
	6   listSerialC                         0x000000010000ece9 get_port_details + 2265
	7   listSerialC                         0x00000001000015c6 sp_get_port_by_name_desc + 678
	8   listSerialC                         0x0000000100002551 list_append + 145
	9   listSerialC                         0x000000010000f398 list_ports + 728
	10  listSerialC                         0x000000010000282c sp_list_ports + 300
	11  listSerialC                         0x0000000100000f7f main + 47
	12  listSerialC                         0x0000000100000f24 start + 52
	13  ???                                 0x0000000000000001 0x0 + 1
)
libc++abi.dylib: terminating with uncaught exception of type NSException
Abort trap: 6
```

danhusan, 2016-12-05:
@facchinm 1.6.13 solved it for me. Thank you very much.

facchinm, 2016-12-06:
@fototakas , one last binary to test (this does not contain the check, so it shouldn't crash at all). Since the bus/address information is not used anywhere, it should be safe to avoid checking it (see https://github.com/arduino/libserialport/commit/6c406689b6018dd0da0acde76b6db1bad9351cb3)
[osx.zip](https://github.com/arduino/Arduino/files/633411/osx.zip)

fototakas, 2016-12-07:
@facchinm Sorry Martino, but still no go:
LIBSERIALPORT_DEBUG=1 ./listSerialC
```
sp: sp_list_ports(0x7fff5fbffa90) called.
sp: Enumerating ports.
sp: Creating matching dictionary.
sp: Getting matching services.
sp: Iterating over results.
sp: Found port /dev/cu.Bluetooth-Incoming-Port.
sp: sp_get_port_by_name_desc(/dev/cu.Bluetooth-Incoming-Port, 0x1001034d0) called.
sp: Building structure for port /dev/cu.Bluetooth-Incoming-Port.
sp: Getting serial port list.
sp: Iterating over results.
sp: Found port /dev/cu.Bluetooth-Incoming-Port.
sp: Found description Bluetooth-Incoming-Port.
sp: get_port_details returning SP_OK.
sp: sp_get_port_by_name_desc returning SP_OK.
sp: Found port /dev/cu.pci-serial22.
sp: sp_get_port_by_name_desc(/dev/cu.pci-serial22, 0x100500078) called.
sp: Building structure for port /dev/cu.pci-serial22.
sp: Getting serial port list.
sp: Iterating over results.
sp: Found port /dev/cu.pci-serial22.
sp: Found description PCI Serial Adapter (22).
sp: get_port_details returning SP_OK.
sp: sp_get_port_by_name_desc returning SP_OK.
sp: Found port /dev/cu.wwan.
sp: sp_get_port_by_name_desc(/dev/cu.wwan, 0x100500080) called.
sp: Building structure for port /dev/cu.wwan.
sp: Getting serial port list.
sp: Iterating over results.
sp: Found port /dev/cu.wwan.
sp: Found description DW5550.
2016-12-06 20:50:24.115 listSerialC[7956:1585862] -[__NSCFString _getValue:forType:]: unrecognized selector sent to instance 0x1001069c0
2016-12-06 20:50:24.120 listSerialC[7956:1585862] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[__NSCFString _getValue:forType:]: unrecognized selector sent to instance 0x1001069c0'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff9a1234f2 __exceptionPreprocess + 178
	1   libobjc.A.dylib                     0x00007fff8d5d473c objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff9a18d1ad -[NSObject(NSObject) doesNotRecognizeSelector:] + 205
	3   CoreFoundation                      0x00007fff9a093571 ___forwarding___ + 1009
	4   CoreFoundation                      0x00007fff9a0930f8 _CF_forwarding_prep_0 + 120
	5   CoreFoundation                      0x00007fff9a034ecf CFNumberGetValue + 287
	6   listSerialC                         0x000000010000ed24 get_port_details + 1860
	7   listSerialC                         0x0000000100001796 sp_get_port_by_name_desc + 678
	8   listSerialC                         0x0000000100002721 list_append + 145
	9   listSerialC                         0x000000010000f3d8 list_ports + 728
	10  listSerialC                         0x00000001000029fc sp_list_ports + 300
	11  listSerialC                         0x000000010000114f main + 47
	12  listSerialC                         0x00000001000010f4 start + 52
)
libc++abi.dylib: terminating with uncaught exception of type NSException
Abort trap: 6

```

facchinm, 2016-12-12:
:sob: As a last chance, I'm blacklisting the WWAN boards till I can get my hands on the real hardware and code a proper fix. If this last binary works without any side effect I'd merge it an release in next IDE update. Thanks for the patience :smile: 
[osx.zip](https://github.com/arduino/Arduino/files/645797/osx.zip)

fototakas, 2016-12-12:
@facchinm Hello Martino!
Now it works, thank you for your support and help. If you will need more testing please let me know.

LIBSERIALPORT_DEBUG=1 ./listSerialC:
```
sp: sp_list_ports(0x7fff5fbffa80) called.
sp: Enumerating ports.
sp: Creating matching dictionary.
sp: Getting matching services.
sp: Iterating over results.
sp: Found port /dev/cu.pci-serial22.
sp: sp_get_port_by_name_desc(/dev/cu.pci-serial22, 0x100500000) called.
sp: Building structure for port /dev/cu.pci-serial22.
sp: Getting serial port list.
sp: Iterating over results.
sp: Found port /dev/cu.pci-serial22.
sp: Found description PCI Serial Adapter (22).
sp: get_port_details returning SP_OK.
sp: sp_get_port_by_name_desc returning SP_OK.
sp: Found port /dev/cu.Bluetooth-Incoming-Port.
sp: sp_get_port_by_name_desc(/dev/cu.Bluetooth-Incoming-Port, 0x100102738) called.
sp: Building structure for port /dev/cu.Bluetooth-Incoming-Port.
sp: Getting serial port list.
sp: Iterating over results.
sp: Found port /dev/cu.Bluetooth-Incoming-Port.
sp: Found description Bluetooth-Incoming-Port.
sp: get_port_details returning SP_OK.
sp: sp_get_port_by_name_desc returning SP_OK.
sp: Found port /dev/cu.wwan.
sp: sp_get_port_by_name_desc(/dev/cu.wwan, 0x100102740) called.
sp: Building structure for port /dev/cu.wwan.
sp: Getting serial port list.
sp: Iterating over results.
sp: get_port_details returning SP_OK.
sp: sp_get_port_by_name_desc returning SP_OK.
sp: sp_list_ports returning SP_OK.
sp: sp_get_port_name(0x100500010) called.
sp: sp_get_port_name returning /dev/cu.pci-serial22.
sp: sp_get_port_by_name_desc(/dev/cu.pci-serial22, 0x7fff5fbffa70) called.
sp: Building structure for port /dev/cu.pci-serial22.
sp: Getting serial port list.
sp: Iterating over results.
sp: Found port /dev/cu.pci-serial22.
sp: Found description PCI Serial Adapter (22).
sp: get_port_details returning SP_OK.
sp: sp_get_port_by_name_desc returning SP_OK.
sp: sp_get_port_usb_vid_pid(0x100403690) called.
sp: sp_get_port_usb_vid_pid returning SP_ERR_ARG: Port does not use USB transport.
sp: sp_free_port(0x100403690) called.
sp: Freeing port structure.
sp: sp_free_port returning.
sp: sp_get_port_name(0x100101bd0) called.
sp: sp_get_port_name returning /dev/cu.Bluetooth-Incoming-Port.
sp: sp_get_port_by_name_desc(/dev/cu.Bluetooth-Incoming-Port, 0x7fff5fbffa70) called.
sp: Building structure for port /dev/cu.Bluetooth-Incoming-Port.
sp: Getting serial port list.
sp: Iterating over results.
sp: Found port /dev/cu.Bluetooth-Incoming-Port.
sp: Found description Bluetooth-Incoming-Port.
sp: get_port_details returning SP_OK.
sp: sp_get_port_by_name_desc returning SP_OK.
sp: sp_get_port_usb_vid_pid(0x100600050) called.
sp: sp_get_port_usb_vid_pid returning SP_ERR_ARG: Port does not use USB transport.
sp: sp_free_port(0x100600050) called.
sp: Freeing port structure.
sp: sp_free_port returning.
sp: sp_get_port_name(0x100101c20) called.
sp: sp_get_port_name returning /dev/cu.wwan.
sp: sp_get_port_by_name_desc(/dev/cu.wwan, 0x7fff5fbffa70) called.
sp: Building structure for port /dev/cu.wwan.
sp: Getting serial port list.
sp: Iterating over results.
sp: get_port_details returning SP_OK.
sp: sp_get_port_by_name_desc returning SP_OK.
sp: sp_get_port_usb_vid_pid(0x100403660) called.
sp: sp_get_port_usb_vid_pid returning SP_ERR_ARG: Port does not use USB transport.
sp: sp_free_port(0x100403660) called.
sp: Freeing port structure.
sp: sp_free_port returning.
sp: sp_free_port_list(0x100102730) called.
sp: Freeing port list.
sp: sp_free_port(0x100500010) called.
sp: Freeing port structure.
sp: sp_free_port returning.
sp: sp_free_port(0x100101bd0) called.
sp: Freeing port structure.
sp: sp_free_port returning.
sp: sp_free_port(0x100101c20) called.
sp: Freeing port structure.
sp: sp_free_port returning.
sp: sp_free_port_list returning.
```

closed 2017-01-19 by facchinm