arduino/Arduino #5212: OSX crashes occasionally when uploading to MKR1000

soundanalogous, 2016-08-06:
Occasionally attempting to upload a sketch to an MKR1000 results in a kernel panic on my Macbook Pro. This happens about half way through the upload process.

I see another user has posted the same issue in the forums a while back: https://forum.arduino.cc/index.php?topic=411252.0

My setup:
MacBook Pro (Retina, 15-inch, Late 2013)
OSX version 10.11.6 (El Capitan) and Arduino 1.6.10
I've also experienced this on older versions of the IDE (maybe back to 1.6.8).

Not sure if this is useful at all, but here is part of the panic report:
```
*** Panic Report ***
panic(cpu 2 caller 0xffffff80075ce6fa): Kernel trap at 0xffffff7f8a4beeb8, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000000000, CR3: 0x00000002ce4c6095, CR4: 0x00000000001627e0
RAX: 0x0000000000000000, RBX: 0xffffff803afb9200, RCX: 0x0000000000005520, RDX: 0x0000000000000001
RSP: 0xffffff81f120bb30, RBP: 0xffffff81f120bb50, RSI: 0x0000000000000002, RDI: 0x0000000000000000
R8:  0xffffff8048bc4000, R9:  0x0000000000000001, R10: 0x0000000000000000, R11: 0x000009d33e607a84
R12: 0x0000000000000000, R13: 0x00000000fffffffd, R14: 0x00000000e00002cd, R15: 0xffffff8044a2f860
RFL: 0x0000000000010286, RIP: 0xffffff7f8a4beeb8, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x0000000000000000, Error code: 0x0000000000000000, Fault CPU: 0x2, PL: 0

Backtrace (CPU 2), Frame : Return Address
0xffffff81f120b7c0 : 0xffffff80074dab52 
0xffffff81f120b840 : 0xffffff80075ce6fa 
0xffffff81f120ba20 : 0xffffff80075ec563 
0xffffff81f120ba40 : 0xffffff7f8a4beeb8 
0xffffff81f120bb50 : 0xffffff8007ab6148 
0xffffff81f120bbc0 : 0xffffff7f88bc23b4 
0xffffff81f120bc00 : 0xffffff7f88bbfc1b 
0xffffff81f120bc30 : 0xffffff8007738c83 
0xffffff81f120bc80 : 0xffffff80077407fd 
0xffffff81f120bcd0 : 0xffffff8007726fa7 
0xffffff81f120bd30 : 0xffffff800771aae3 
0xffffff81f120bda0 : 0xffffff800771c759 
0xffffff81f120be50 : 0xffffff8007986980 
0xffffff81f120bec0 : 0xffffff80079824b0 
0xffffff81f120bf20 : 0xffffff8007986624 
0xffffff81f120bf60 : 0xffffff8007a28701 
0xffffff81f120bfb0 : 0xffffff80075ecd66 
      Kernel Extensions in backtrace:
         com.apple.iokit.IOSerialFamily(11.0)[854221E2-E1EB-361A-98C6-DC6B7E1EB0F4]@0xffffff7f88bbf000->0xffffff7f88bccfff
         com.apple.driver.usb.cdc.acm(5.0)[C697B384-9D04-3983-AD3B-A83C3E19F133]@0xffffff7f8a4bb000->0xffffff7f8a4c8fff
            dependency: com.apple.driver.usb.cdc(5.0.0)[204EA973-076B-3517-93BF-9057AD3DDBD6]@0xffffff7f89af5000
            dependency: com.apple.driver.usb.networking(5.0.0)[70876950-E9CC-313C-A239-5C1E5CA936BF]@0xffffff7f89aed000
            dependency: com.apple.iokit.IOSerialFamily(11)[854221E2-E1EB-361A-98C6-DC6B7E1EB0F4]@0xffffff7f88bbf000
            dependency: com.apple.iokit.IOUSBHostFamily(1.0.1)[79D250A3-843A-3750-BE64-A252CF17A148]@0xffffff7f886de000
            dependency: com.apple.driver.usb.AppleUSBHostCompositeDevice(1.0.1)[DC814FD4-7773-3054-8D2C-AFA8E6CA034A]@0xffffff7f88a2a000

BSD process name corresponding to current thread: bossac
```

facchinm, 2016-08-10:
Hi Jeff, 
it looks like `bossac` is smashing something in cdc driver, this could be caused by the "recent" changes in OSX USB layer which are causing other problems with other boards (difficulties in enumeration and upload).
Are you able to reproduce this consistently (like, let's say, on every 10th upload) or is it a once in a while event?

soundanalogous, 2016-08-11:
It's somewhat random. For example I just tried uploading over 20 times and no issue. Other times it's every 2 or 3 times I try to upload. May have something to do with other events running on the system that are not running currently. I had suspected Google Drive sync as interfering at one point. Will try to gather more data.

sandeepmistry, 2016-12-12:
Closing this for now due to lack of feedback.

closed 2016-12-12 by sandeepmistry