arduino/Arduino #5214: [Website] Faults in Tutorial's Sample code

Enigma-TS, 2016-08-06:
IMHO the code on this page has a few issues:
     https://www.arduino.cc/en/Tutorial/DimmingLEDs

Is this the place to tell someone about this?
If not, where would you suggest?

I have a commented version of the existing Code, describing the issues.
I also have an alternative that could be used to substitute for it (Assuming someone is interested).

matthijskooijman, 2016-08-06:
Here is a good place to provide feedback about examples and documentation. A proposal for a fix is also welcome!

Enigma-TS, 2016-08-07:
I was trained that feedback to the Author of something doesn't necessarily mean ANY changes should/need/must be made - Implementation of suggestions is SOLELY at the discretion of the Author.  See:
-     http://www.dorsethouse.com/pdf/Dorset-House-hdbk.pdf

Accordingly, feel free to bin any and all of my comments.

OK, with that out of the way...
I have only been a member for about 30 minutes longer than the age of this post (And writing code for the Arduino for about 7 days - Coding in other languages for longer), so here goes...

1  -  [DimmingLEDs.ino.txt](https://github.com/arduino/Arduino/files/405305/DimmingLEDs.ino.txt)

2  -  [DimmingLEDs_Output.txt](https://github.com/arduino/Arduino/files/405306/DimmingLEDs_Output.txt)

3  -  [LED Cross-Fader.ino.txt](https://github.com/arduino/Arduino/files/405307/LED.Cross-Fader.ino.txt)

4  -  [LED Cross-Fader_Output.txt](https://github.com/arduino/Arduino/files/405308/LED.Cross-Fader_Output.txt)

I trust the 4 files will be self explanatory.  Otherwise, please let me know.

Regards from 'Down Under',   Enigma-TS.

Enigma-TS, 2016-08-11:
I have more improvements, if you want them.

What is the Process for reviewing things like this?
And, who closes this thread?

_
Regards from 'Down Under', Enigma-TS.

closed 2016-08-16 by Enigma-TS